YosysHQ/icestorm #1: Problem compiling

habiloid, 2015-07-29:
Hi
When running make -j$(nproc)

I get:

make -C icebox
make[1]: Entering directory '/home/habiloid/icestorm/icebox'
python icebox_chipdb.py > chipdb-1k.new
python icebox_chipdb.py -8 > chipdb-8k.new
Traceback (most recent call last):
  File "icebox_chipdb.py", line 250, in <module>
    print_tile_nonrouting_bits("logic", ic.logic_tiles.keys()[0])
TypeError: 'dict_keys' object does not support indexing
Makefile:6: recipe for target 'chipdb-1k.txt' failed
make[1]: *** [chipdb-1k.txt] Error 1
make[1]: *** Waiting for unfinished jobs....
Traceback (most recent call last):
  File "icebox_chipdb.py", line 250, in <module>
    print_tile_nonrouting_bits("logic", ic.logic_tiles.keys()[0])
TypeError: 'dict_keys' object does not support indexing
Makefile:10: recipe for target 'chipdb-8k.txt' failed
make[1]: *** [chipdb-8k.txt] Error 1
make[1]: Leaving directory '/home/brett/BUILDS/COMPILATIONS/icestorm/icebox'
Makefile:3: recipe for target 'all' failed
make: *** [all] Error 2

Any help would be appreciated

Ta

cliffordwolf, 2015-07-29:
What version of Python are you using? (output of `python --version`)

habiloid, 2015-07-29:
3.4.3

cliffordwolf, 2015-07-30:
Fixed in 4c1335f: Those python scripts are written in python 2, not python 3. Now the script call the "python2" binary instead of relying on "python" being a python 2 interpreter. Thanks for bringing this problem to my attention.

closed 2015-07-30 by cliffordwolf