YosysHQ/icestorm #4: OSX problems

cseed, 2015-08-19:
There is no python2 installed on OSX.  I created a link on my own machine, but that's not possible for people without admin rights (e.g. running on travis-ci).

qbit, 2015-08-19:
the latest commit allows you to specify the location of a python interpreter. Do something like this:

```
env PYTHON=python make
```

cseed, 2015-08-19:
`/usr/bin/python2` is still hardcoded in the various python scripts (e.g., icebox/icebox_vlog.py).

qbit, 2015-08-19:
Ahh - I see.

cliffordwolf, 2015-08-20:
Also: python2 and python3 are different languages. I have converting the icestorm scripts to python3 on my todo list anyways. (That's why I added python3 to the Ubuntu install instructions a while back.) Maybe I'll even do it today..

cliffordwolf, 2015-08-22:
Icestorm is now using python 3: f6f96a26e036e56577ad5a1bfec78fd06f419ca0 498a7acbe2a708148ed7cbf6934debb1193f656d b8494dae8cef3767d68ddff814c3f2b0f8a1a2c2

closed 2015-08-22 by cliffordwolf