YosysHQ/icestorm #16: [iceprog] It takes more than 30 seconds to download the bistream in some computers

Obijuan, 2015-12-08:
I have tested in three different laptops. What I got is:

$ time sudo iceprog scicad1.bin
init..
cdone: high
reset..
cdone: low
flash ID: 0x20 0xBA 0x16 0x10 0x00 0x00 0x23 0x12 0x67 0x21 0x23 0x00 0x21 0x00 0x43 0x04 0x11 0x11 0x5F 0x7D
file size: 32220
erase 64kB sector at 0x000000..
programming..
reading..
VERIFY OK
cdone: high
Bye.

real	0m33.554s
user	0m0.016s
sys	0m0.056s

It takes 33 seconds to perform the programming.  But in other computers it just takes 3 seconds.

I made one test in two laptops with ubuntu 15.10. Same kernel, same linux distro. Laptop A took 30 seconds and Laptop B only 3sec

Laptop A:  High-end laptop: i7, 8GB RAM
Laptop B: Low-end laptop: intel core duo, 4GB RAM (five years old)

Any help or suggestions on test to perform are welcome

Obijuan, 2015-12-08:
I have seen the iceprog code.

In the function flash_wait() there is a 250ms delay (at the end):

usleep(250000);

I have changed it to 10ms:

usleep(10000);

The result is that now it only takes 4 sec on the Laptop A:

$ time sudo ./iceprog scicad1.bin 
init..
cdone: high
reset..
cdone: low
flash ID: 0x20 0xBA 0x16 0x10 0x00 0x00 0x23 0x12 0x67 0x21 0x23 0x00 0x21 0x00 0x43 0x04 0x11 0x11 0x5F 0x7D
file size: 32220
erase 64kB sector at 0x000000..
programming.............
reading..
VERIFY OK
cdone: high
Bye.

real	**0m3.495s**
user	0m0.004s
sys	0m0.088s

I will test on the laptop B and tell you the result

cliffordwolf, 2015-12-08:
Thanks for reporting this issue.

I have had reports of that before, but unfortunately I got no answers to my followup questions so far and I cannot reproduce this problem on any of my machines..

I have some questions:

Are you using the icestick directly in the USB port or are you using a USB extension cable? Does the problem go away if you use the icestick directly in the USB port?

Does the behavior change on either machine if you put a USB hub between the icestick and the host?

Can you run `strace -o logfile.txt -r iceprog scicad1.bin` on both machines and post the logfiles created? (Upload somewhere and post the links here, or send an email to clifford@clifford.at.)

On the machine where it takes 33 seconds, are there any relevant entries in `dmesg`?

cliffordwolf, 2015-12-08:
> I have changed it to 10ms:
> usleep(10000);

Ok, that makes sense of course.

The 250 ms are a copy&paste error. I've now set to to 1 ms.

closed 2015-12-08 by cliffordwolf

Obijuan, 2015-12-08:
Great! thanks!  :-)   I`ve also tested the icestick connected directly to the USB port: I got the same result than using an USB hub and an extension cable: 33sec (just for having more information). So the bug seems to be that copy&paste error

Thanks a lot Clifford!  ;-)

carlosgs, 2015-12-08:
I just tested this fix, turning the value to 1ms, and it does bring the programming time down from 33.707s to 0.387s in my i5 laptop (with Fedora). Great find Juan!!! :+1:

Obijuan, 2015-12-08:
Value set to 1ms: It takes 2.1sec. Awesome!