YosysHQ/icestorm #29: Problem building on Mac OS X 10.9.5

gc179, 2016-02-19:
Hello,

The make fails with:

[Geoffreys-MacBook-Pro-2:fpga/tools/icestorm] geoffc% make
/Applications/Xcode.app/Contents/Developer/usr/bin/make -C icebox
make[1]: Nothing to be done for `all'.
/Applications/Xcode.app/Contents/Developer/usr/bin/make -C icepack
make[1]: Nothing to be done for `all'.
/Applications/Xcode.app/Contents/Developer/usr/bin/make -C iceprog
clang -MD -O0 -ggdb -Wall -std=c99 -I/usr/local/include   -c -o iceprog.o iceprog.c
In file included from iceprog.c:27:
/usr/local/include/ftdi.h:20:10: fatal error: 'usb.h' file not found
#include <usb.h>
         ^
1 error generated.
make[1]: *** [iceprog.o] Error 1
make: *** [all] Error 2

I'm not sure which package I might to supply the usb.h file. Has anyone else run across this one?

Thanks.

Geoff

jacobrosenthal, 2016-02-19:
makes for me on osx. Presumably you're missing libusb or libusb-compat (I dont know which as I have both installed)

Grab them with homebrew
`brew install libusb libusb-compat`

gc179, 2016-02-20:
Thanks! It turned out I had libusb and libusb-compat but libusb-compat was Somehow Messed Up (tm).

A brew uninstall/install got it to build, and I got the whole toolchain built and made the 'rot' example.

Now if Newark would just ship my iCEstick! :)

Geoff

closed 2016-02-20 by gc179

silardg, 2023-07-03:
I would reopen this as it is still giving this error. 
[image]