YosysHQ/icestorm #33: iceprog requires additional compiler flags

resetnow, 2016-02-28:
On Fedora 23, iceprog cannot be built because it requires additional linker and compiler flags for libftdi. I [tried](https://github.com/resetnow/icestorm/commit/2792a4277595c0086d6945d89115889db46be8f1) to fix this, but it seems like different GNU/Linux distributions have different library names for libftdi. I'm not quite sure how to process this in makefile.

Fedora 23:
```
[user@fedora]$ pkg-config --libs --cflags libftdi1
-I/usr/include/libftdi1 -I/usr/include/libusb-1.0 -lftdi1 -lusb-1.0 
```

Ubuntu 12.04:
```
user@ubuntu:~$ pkg-config --libs --cflags libftdi
 -lftdi  
```

As a side note, maybe it would be more convenient to have a more flexible build system, like cmake or premake? With premake, for example, it would be easier to provide configurations for variety of compilers for different platforms. [Here](https://gist.github.com/resetnow/98aa71204a8f47064452) is a quick example for icestorm.

cliffordwolf, 2016-02-29:
I think I fixed it. Please test.

resetnow, 2016-02-29:
It works.

closed 2016-02-29 by resetnow