YosysHQ/icestorm #34: Chip DB for the iCE40-HX1K should report that there is no PLL in the VQ100 package.

cfib, 2016-03-18:
The following design should not be place-able on the iCE40-HX1K VQ100
[pll.tar.gz](https://github.com/cliffordwolf/icestorm/files/179405/pll.tar.gz)

http://www.latticesemi.com/~/media/LatticeSemi/Documents/ApplicationNotes/IK/iCE40sysCLOCKPLLDesignandUsageGuide.pdf?document_id=47778

cliffordwolf, 2016-04-16:
Sorry for keeping you waiting for so long. Yes, you are absolutely right. Unfortunately there is no interface yet between icestorm and arachne-pnr to communicate what PLLs to use in what package (because it is the same chip, the difference is just if the PLLVCC/PLLGND pads are bonded to package pins). Right now arachne-pnr is simply using all PLLs on the chip, not knowing which PLL cannot be used in which package because of missing PLLVCC/PLLGND pins. I'll extend the chipdb file format accordingly the next time I do work on icestorm and keep this issue open until then.

cliffordwolf, 2016-05-15:
I have now added this information to the IceStorm ChipDB file (see commit dcf4e6c26cf86a59128d214e0ac6d6ef3fcf11d1). However, arachne-pnr does not yet use it. I have now created an issue for that in the arachne-pnr github repo:

https://github.com/cseed/arachne-pnr/issues/34

I'm now closing this issue here as the IceStorm side of things is now implemented.

closed 2016-05-15 by cliffordwolf