YosysHQ/icestorm #36: Has LVDS out been tested?

peepo, 2016-04-15:
Clifford,

Has LVDS out been tested**?
code sample: [lvds.txt](https://github.com/cliffordwolf/icestorm/files/220507/lvds.txt)

simple LVDS code builds with no errors reported:

using iCEcube2 runs as expected
    Total number of logic levels: 12
    Total path delay: 14.37 ns (69.58 MHz)

on iCEstorm does not run as expected
    Total number of logic levels: 22
    Total path delay: 23.95 ns (41.76 MHz)

**I only found reference to LVDS in:
http://www.clifford.at/icestorm/io_tile.html

kallisti5, 2016-04-15:
Ah.. No wonder I never got my lvds code to do anything.

cliffordwolf, 2016-04-16:
Please post a complete synthesizable example, including pcf file.

There is no special hardware support in iCE40 devices for LVDS _output_ (the dedicated LVDS hardware is for input only), so it is not clear to me what you think is not supported or not tested.

Quoting Lattice TN1253 "Using Differential I/O (LVDS, Sub-LVDS) in iCE40 LP/HX Devices":

> For some differential I/O standards, such as LVDS, the output driver is actually a current source. On iCE40 FPGAs, however, differential outputs are constructed using a pair of single-ended PIO pins as shown in Figure 3, and an external resistor network consisting of three resistors. Because differential outputs are built from two single-ended LVCMOS outputs, differential outputs are available in any I/O bank.

peepo, 2016-04-16:
Clifford, 
Are you able to post "a complete synthesizable example" using
.PIN_TYPE = 6'b011100; that you believe to work?
NB: does it build similarly on iCEcube2 and iCEstorm?

as I wrote you by email, in order to create a reduced testcase or "a complete synthesizable example" that clearly identifies LVDS as the issue, I need further guidance:

This might be for instance be:
A way for me to identify the differences between the examples synthesised by iCEstorm and iCEcube2.

Clearly the difference between 22 logic levels and 12 may be significant, 
I need to understand how to assess this in order to define the issue more clearly for both of us.

regards

Jonathan

cliffordwolf, 2016-04-16:
> Clifford, 
Are you able to post "a complete synthesizable example" using
.PIN_TYPE = 6'b011100; that you believe to work?
NB: does it build similarly on iCEcube2 and iCEstorm?

First of all: It is custom, when writing a bug report, to provide an example that demonstrates the alleged bug, not to demand proof from the original developer that the (vaguely described) bug does not exists. This is ridiculous, a waste of my time, and extremely inconsiderate. Especially considering that you have to have some kind of example already, but for some reason you have decided to only post a fraction of it, and now refuse to post a complete example, wasting my time because now I have to try to reproduce what's going on on your side, without you showing me what you are trying to do.

However, here is an example that does work on the iCEstick, with Yosys+Arachne-pnr as well as with iCEcube:

```
module top (
        input  clk,
        output LED1,
        output LED2,
        output LED3,
        output LED4,
        output LED5
);

        localparam LOG2DELAY = 22;

        reg [LOG2DELAY-1:0] counter = 0;
        reg slowclock = 0;
        reg outval = 0;

        always@(posedge clk) begin
                if (!counter)
                        slowclock <= !slowclock;
                counter <= counter + 1;
        end

        always@(posedge slowclock) 
                outval <= !outval;

        SB_IO #(
                .PIN_TYPE(6'b 0111_00)
        ) test_P (
                .PACKAGE_PIN(LED5),
                .OUTPUT_CLK(slowclock),
                .D_OUT_0(outval)
        );

        assign {LED1, LED2, LED3, LED4} = 0;
endmodule
```

Here is the PCF file that goes with it:

```
set_io LED1 99
set_io LED2 98
set_io LED3 97
set_io LED4 96
set_io LED5 95
set_io clk 21
```

This is a variation of the `examples/icestick/` example bundled with project icestorm. I simply used to makefile from that example.

As I have now proven to you that this indeed works, and you have refused to post an example that does not work, I am now closing this issue. This is the first and last time I do this. In the future, if you think you have found a bug, but refuse to post example code, I will simply close the issue without any further comment. I have only limited time and will not spend it on BS like this.

> Clearly the difference between 22 logic levels and 12 may be significant,

Clearly this is an independent problem and clearly I can't do anything about it when you do not post your test case.

I appreciate your interest in project icestorm, but do not waste my time like this ever again.

Don't get me wrong: It is OK to report an issue that turns out the be a problem on your side. Most reported issues are. But this "I'm not going to post my example -- you have to post an example that proves that it is working" crap is unacceptable, and if you try to pull it ever again, there is a certain chance that I will simply ignore all further communications from you. I hope this is clear now.

Have a good day.

closed 2016-04-16 by cliffordwolf

peepo, 2016-04-16:
Clifford

your example is not an LVDS pair, so may not be related to this bug.
-- differential being the keyword.

I do understand we all have bad hair days, but really you need to take a time out.

best

Jonathan

cliffordwolf, 2016-04-16:
As I wrote before:

1) THERE IS NO HARDWARE SUPPORT IN ICE40 FOR LVDS OUTPUT! IT IS JUST TWO PIOS! READ THE PARAGRAPH I QUOTED FROM THE LATTICE DOCUMENTATION!

2) EITHER PROVIDE AN EXAMPLE FOR WHAT YOU THINK IS NOT WORKING OR LEAVE ME ALONE!

> I do understand we all have bad hair days, but really you need to take a time out.

Clearly you don't seem to understand that I asked _repeatedly_ for information you decline to provide. That is not me having a "bad hair day", that you wasting my time for no good reason. I can't make you understand that, but I can lock this conversation. Go waste someone else's time!