YosysHQ/icestorm #39: recipe for target 'iceprog.o' failed

drom, 2016-05-16:
```sh
make -C iceprog
make[1]: Entering directory '/home/drom/work/github/cliffordwolf/icestorm/iceprog'
clang -MD -O0 -ggdb -Wall -std=c99 -I/usr/local/include    -c -o iceprog.o iceprog.c
iceprog.c:27:10: fatal error: 'ftdi.h' file not found
#include <ftdi.h>
         ^
1 error generated.
<builtin>: recipe for target 'iceprog.o' failed
make[1]: *** [iceprog.o] Error 1
make[1]: Leaving directory '/home/drom/work/github/cliffordwolf/icestorm/iceprog'
Makefile:4: recipe for target 'all' failed
make: *** [all] Error 2
```

drom, 2016-05-16:
I have fixed `ftdi.h` error with installation of `libftdi1-devel` package. But now I am getting this:
```
...
make -C iceprog
make[1]: Entering directory '/home/drom/work/github/cliffordwolf/icestorm/iceprog'
clang -MD -O0 -ggdb -Wall -std=c99 -I/usr/local/include -I/usr/include/libftdi1 -I/usr/include/libusb-1.0    -c -o iceprog.o iceprog.c
iceprog.c:58:2: warning: implicitly declaring library function 'exit' with type 'void (int) __attribute__((noreturn))'
        exit(1);
        ^
iceprog.c:58:2: note: include the header <stdlib.h> or explicitly provide a declaration for 'exit'
iceprog.c:377:16: warning: implicit declaration of function 'strtol' is invalid in C99 [-Wimplicit-function-declaration]
                        read_size = strtol(optarg, &endptr, 0);
                                    ^
2 warnings generated.
```

cliffordwolf, 2016-05-16:
Commit 1b7253ade29f89657a0f78d3b3659bc52ad0bac0 should fix those warning.

closed 2016-05-16 by cliffordwolf

jhonny-ramirez-salamancasolutions-com, 2017-11-27:
You need to install the **libftdi-dev** on Ubuntu 16.04.2 LTS
`sudo apt-get install libftdi-dev`