YosysHQ/icestorm #41: JSON timing report and wire delay

drom, 2016-05-16:
Some important information in timing report present in textual format and absent in JSON.
  1) wire detay (or cell delay?) on of those
  2) extended wire name like `$auto$alumacc.cc:470:replace_alu$9.C[3]`

text:
```
        lc40_1_2_1 (LogicCell40) carryin -> carryout: 0.126 ns
     1.987 ns net_4014 ($auto$alumacc.cc:470:replace_alu$9.C[2])
        lc40_1_2_2 (LogicCell40) carryin -> carryout: 0.126 ns
     2.113 ns net_4020 ($auto$alumacc.cc:470:replace_alu$9.C[3])
```

json:
```js
    { out_net: "net_4014", cell: "lc40_1_2_1", cell_type: "LogicCell40", cell_in_port: "carryin", cell_out_port: "carryout", delay_ns: 1.987, },
    { out_net: "net_4020", cell: "lc40_1_2_2", cell_type: "LogicCell40", cell_in_port: "carryin", cell_out_port: "carryout", delay_ns: 2.113, },
```

cliffordwolf, 2016-05-16:
1. The cells with ports "I" and "O" are interconnect resources. They are the "wires" if you will

2. Symbolic wire names are now included in the JSON format as "net" property. The "out_net" property has been renamed to "hwnet".

closed 2016-05-16 by cliffordwolf

drom, 2016-05-16:
Great. My concern was that in text report I can see 2 time intervals per cell ans 1 number in JSON report. Like in case above:

text: ```0.126 ns,  1.987 ns, 0.126 ns, 2.113 ns```

JSON: ```1.987 ns, 2.113 ns```

cliffordwolf, 2016-05-16:
The delay in the json output is the accumulative delay, i.e. the numbers on the left margin of the text output. In both cases it's 1.987 ns and 2.113 ns. The numbers in the other lines of the text output are the added delay for this cell (e.g. 1.987 ns + 0.126 ns = 2.113).

The text output does only contain an extra line with the accumulative delay at points like logic cell outputs. The json format only contains accumulative delay, not the per-cell added delay. E.g. text format:

```
    23.364 ns net_25682 ($abc$52278$n3452)
        t6834 (LocalMux) I -> O: 0.330 ns
        inmux_7_10_29800_29843 (InMux) I -> O: 0.260 ns
        lc40_7_10_2 (LogicCell40) in3 -> lcout: 0.316 ns
    24.268 ns net_25679 ($abc$52278$n3424)
        odrv_7_10_25679_5146 (Odrv12) I -> O: 0.540 ns
        t6835 (LocalMux) I -> O: 0.330 ns
        inmux_9_10_37284_37315 (InMux) I -> O: 0.260 ns
        lc40_9_10_1 (LogicCell40) in0 -> lcout: 0.449 ns
    25.846 ns net_33612 ($abc$52278$n619)
```

And corresponding json output:

```
    { "net": "$abc$52278$n3452", "hwnet": "net_25682", "cell": "lc40_7_10_5", "cell_type": "LogicCell40", "cell_in_port": "in0", "cell_out_port": "lcout", "delay_ns": 23.364 },
    { "hwnet": "seg_7_10_local_g0_5_29800", "cell": "t6834", "cell_type": "LocalMux", "cell_in_port": "I", "cell_out_port": "O", "delay_ns": 23.693 },
    { "hwnet": "net_29843", "cell": "inmux_7_10_29800_29843", "cell_type": "InMux", "cell_in_port": "I", "cell_out_port": "O", "delay_ns": 23.953 },
    { "net": "$abc$52278$n3424", "hwnet": "net_25679", "cell": "lc40_7_10_2", "cell_type": "LogicCell40", "cell_in_port": "in3", "cell_out_port": "lcout", "delay_ns": 24.268 },
    { "hwnet": "net_5146", "cell": "odrv_7_10_25679_5146", "cell_type": "Odrv12", "cell_in_port": "I", "cell_out_port": "O", "delay_ns": 24.808 },
    { "hwnet": "seg_9_10_local_g1_0_37284", "cell": "t6835", "cell_type": "LocalMux", "cell_in_port": "I", "cell_out_port": "O", "delay_ns": 25.138 },
    { "hwnet": "net_37315", "cell": "inmux_9_10_37284_37315", "cell_type": "InMux", "cell_in_port": "I", "cell_out_port": "O", "delay_ns": 25.398 },
    { "net": "$abc$52278$n619", "hwnet": "net_33612", "cell": "lc40_9_10_1", "cell_type": "LogicCell40", "cell_in_port": "in0", "cell_out_port": "lcout", "delay_ns": 25.846 },
```