YosysHQ/icestorm #42: icetime, from json support, bug: std::out_of_range

nononame, 2016-05-16:
Hi,

From the commit https://github.com/cliffordwolf/icestorm/commit/bafd99bd841a561e07092f3e515b77f5cebfd567 , icetime terminates with an 'std::out_of_range' exception (tested on an hx8k asc file).
The exception is emited by map::at in icetime.cc line [863](https://github.com/cliffordwolf/icestorm/blob/1b7253ade29f89657a0f78d3b3659bc52ad0bac0/icetime/icetime.cc#L863)

Simon.

cliffordwolf, 2016-05-16:
Sigh. As you can probably guess, it works with the hx8k asc files I have here.

Please post the .asc file you have problems with or I can't do anything about that..

Point 1) in the [Yosys F.A.Q.](http://www.clifford.at/yosys/faq.html) is on how to write a good bug report. I quote:

> As a golden rule: Always consider that someone reading your bug report only has the information you provided. Before clicking the send button on your bug report and/or support request, double check that you have provided all the information necessary to easily reproduce the problem you are seeing.

nononame, 2016-05-16:
Sure, I provide more information:

It seems that it is the same for hx8k and hx1k. It is specific to one of my projects that emits this exception.

Version of my tools:
Yosys 0.6+107 (git sha1 864eead, clang 3.5.0-10 -fPIC -Os)
arachne-pnr 0.1+154+2 (git sha1 6b83364, g++ 4.9.2-10 -O2)
icestorm (bug from commit https://github.com/cliffordwolf/icestorm/commit/bafd99bd841a561e07092f3e515b77f5cebfd567 to https://github.com/cliffordwolf/icestorm/commit/1b7253ade29f89657a0f78d3b3659bc52ad0bac0)

I have reproduced this bug on a 32-bit Jessie Debian and on a 64-bit Jessie Debian.

Arachne-pnr output for device hx8k and hx1k:
https://www.le-huit.fr/ephemeral/hx8k.asc
https://www.le-huit.fr/ephemeral/hx1k.asc

Reproducing the bug:

~~~ {.bash}
$ ./icetime -d hx1k hx1k.asc
// Reading input .asc file..
// Reading 1k chipdb file..
// Creating timing netlist..
// Timing estimate: 7.59 ns (131.78 MHz)
terminate called after throwing an instance of 'std::out_of_range'
  what():  map::at
Abandon
~~~

~~~ {.bash}
$ ./icetime -d hx8k hx8k.asc
// Reading input .asc file..
// Reading 8k chipdb file..
// Creating timing netlist..
// Timing estimate: 7.95 ns (125.85 MHz)
terminate called after throwing an instance of 'std::out_of_range'
  what():  map::at
Abandon
~~~

The design contains:

  - a pll (thanks for your recent tool icepll, very useful)
  - a simple spi interface and some led outputs.

If it is useful I can provide the source code.

I would help you but I am not a C++ developer.

Thanks to have free FPGAs !

cliffordwolf, 2016-05-16:
Thanks for the detailed information. I've fixed the bug in commit 4fbc8d20b65952eeef64a2e4c54b9b0ddfb7de07.

Currently there is no complete timing model for e.g. the PLL control signals such as LOCK. In your case the longest path originates from a PLL LOCK output, which triggered the bug. Designs that have their longest path originate from a FF or IO pin (which is of course the case in the vast majority of cases) are not affected.

closed 2016-05-16 by cliffordwolf

nononame, 2016-05-16:
Perfect, it is fixed.

After test, there was no problem if PLL LOCK is just used as FFs reset.
The bug happened only when PLL LOCK is a combinatorial or FF input but it is now fixed.

Thanks!