YosysHQ/icestorm #56: icebram: Parser not fully compatble with that of $readmemh

zeldin, 2016-09-19:
For icebram to work as a drop-in replacement for incuding the hexfiles directly with $readmemh, it should accept the same hexfile syntax.  The differences I have found:

* icebram requires one word per line.  $readmemh accepts multiple words per line
* icebram does not accept horizontal whitespace in the file.  $readmemh accepts horizontal whitespace in the file
* icebram requires the file to be padded to the size of the BRAM.  $readmemh accepts any size <= the size of the BRAM.

cliffordwolf, 2016-10-02:
Off the top of my head, a few more things that are part of the $readmemh file format that aren't supported by icebram:

- Comments (`// ...` and `/* ... */`)
- Addresses (`@<hexaddr>`)
- `x` and `z` nibbles (interpret as zero)

I actually never intended to support the complete $readmemh file format in icebram, only a small subset. But I can see how it might be useful. Feel free to create a pull request that adds support for some more $readmemh features to icebram. I'll leave this issue open until one of us finds the time to write that code..

piranna, 2017-04-16:
Can this be clossed?

closed 2017-05-04 by cliffordwolf