YosysHQ/icestorm #62: Strange behaviour with unused wire/reg

folknology, 2016-12-21:
Below Verilog does not work (only 2 leds come on), but is fixed when I uncomment the unused wire 'idle' in module debounce. I may be missing something obvious of course...

`
module button(input clk, input [1:0] buttons, output [3:0] led);
	assign led = leds;

	reg [3:0] leds = 4'b0;
	wire on;
	wire off;
	wire pressed;

	debounce dbon(.clk(clk),.button(buttons[0]),.state(on));
	debounce dboff(.clk(clk),.button(buttons[1]),.state(off));

	assign pressed = on | off;

	always @(posedge pressed) 
		leds <= (on) ? 4'b1111 :
					(off) ? 4'b0000 : leds;

endmodule

module debounce(input clk, input button, output reg state);
	//wire idle;
	reg sync;
	reg [15:0] count;

	always @(posedge clk) begin
		sync <= ~button;
		count <= (state == sync) ? 0 : count + 16'b1;
		if(&count) 
			state <= ~state;
	end
endmodule
`
IceStorm vers:
Yosys 0.5+460 (git sha1 45af4a4, clang 3.4-1ubuntu3 -fPIC -Os)
arachne-pnr 0.1+136+0 (git sha1 1a4fdf9, g++ 4.8.4-2ubuntu1~14.04.1 -O2)

cliffordwolf, 2016-12-22:
This is a race condition:

```
 	assign pressed = on | off;

	always @(posedge pressed) 
		leds <= (on) ? 4'b1111 : (off) ? 4'b0000 : leds;
```

When e.g. `on` goes high, what arrives first at the flip-flop: the rising edge on `pressed` or the rising edge on `on`?

The presence of the "unused wire `idle`" will not cause any logical difference, but it might change things like auto-generated net names which in turn will change placement, which explains why the race is resolved differently for different LEDs in the two cases.

There is no need to even have a `pressed` signal. Just do this:

```
	always @(posedge clk) 
		leds <= (on) ? 4'b1111 : (off) ? 4'b0000 : leds;
```

Alternatively, model a proper SR-FF if you really don't want to use a clock here:

```
	always @*
		if (on)
			leds = 4'b1111;
		else if (off)
			leds = 4'b0000;
```

closed 2016-12-22 by cliffordwolf

folknology, 2016-12-22:
The trouble with the latter is that I get the implied latch error:
`chip.blif:285: fatal error: unknown model `$_DLATCH_P_'
make: *** [chip.bin] Error 1
`

And although the former is Ok for the example (sticky state 2 momentary buttons) I also use a similar construct to control the sequence of lighting 4 rotating leds:

```Verilog
assign pressed = right | left;

	always @(posedge pressed) 
		leds <= (right) ? {leds[0],leds[3:1]) :
                        (left) ? {leds[2:0],leds[3]} : leds;
```
And this obviously wouldn't work
```Verilog
assign pressed = right | left;

	always @(posedge clk) 
		leds <= (right) ? {leds[0],leds[3:1]) : 
                        (left) ? {leds[2:0],leds[3]} : leds;
```

folknology, 2016-12-22:
Also 

> The presence of the "unused wire idle" will not cause any logical difference, but it might change things like auto-generated net names which in turn will change placement, which explains why the race is resolved differently for different LEDs in the two cases.

Is interesting so unused wires and registers are not pruned in a first pass or something

cliffordwolf, 2016-12-22:
> The trouble with the latter is that I get the implied latch error:

Update Yosys. Yosys 0.5+460 is a bit old. Current version is Yosys 0.7+47.

> I was assuming that the `assign pressed = on | off` would introduce a gate/Lut delay

Sure. But often routing delays are dominant over gate delays. The rising edge on `on` might still arrive later at the FF than the rising edge on `pressed`.

> so unused wires and registers are not pruned in a first pass?

There are pruned. But not in a first pass. Most unused logic/wires is a bit more complicated than this example and its only a bit later in the flow when all unused resources can easily be identified.

> And this obviously wouldn't work

```
        assign pressed = right | left;

	always @(posedge clk) 
		leds <= (right) ? {leds[0],leds[3:1]) : 
                        (left) ? {leds[2:0],leds[3]} : leds;
```

But this would:

```
        reg right_q, left_q;

        always @(posedge clk) begin
		leds <= (!right_q && right) ? {leds[0],leds[3:1]) : 
                        (!left_q && left) ? {leds[2:0],leds[3]} : leds;
                right_q <= right;
                left_q <= left;
        end
```

folknology, 2016-12-22:
Thanks for the tips, I need to keep more up to date with the iceStorm versions, I have now updated. BTW is there a way to turn on inferred latch warnings with yosys, that would be a rather helpful feature?

cliffordwolf, 2016-12-22:
Just grep for `Latch inferred for signal` messages in the logfile (they are created by the `proc_dlatch` pass).