YosysHQ/icestorm #72: icetime does not accept valid net names for timing analysis

madscientist159, 2017-02-26:
When attempting to generate a timing report for any given global clock net names (verified to exist in the corresponding arachne-pnr output) icestorm always returns:

```
icetime -m -d hx8k -p test.pcf -P ct256 -T secondary_clock test.int
// Reading input .pcf file..
// Reading input .asc file..
// Reading 8k chipdb file..
// Creating timing netlist..

icetime topological timing analysis report
==========================================

Info: max_span_hack is enabled: estimate is conservative.

Report for secondary_clock:
--------------------------------------

Net not found: secondary_clock```

Yet the symbol exists:

```
.sym 6 secondary_clock
```

Is this a bug?  There is next to no documentation on icetime for anything other than generating an absolute worst case timing estimate for the entire design.

Thanks!

madscientist159, 2017-02-26:
Nearest I can tell I need to pass a net_<number> hand-picked from the output of

```
icetime -m -d hx8k -p test.pcf -P ct256 -o test.v test.int
```

icetime is then able to look up (resolve) the correct net names from the original source, however it is generating bogus (low) timing delays due to mysteriously missing chunks of logic.  If it makes any difference, the clock in question is a global clock using a SB_GB_IO block in the Verilog source.

madscientist159, 2017-02-26:
A bit more data...the clock in question is net_6 in the Verilog netlist dump, but icetime still refuses to generate a report for it:

```
// Reading input .pcf file..
// Reading input .asc file..
// Reading 8k chipdb file..
// Creating timing netlist..

icetime topological timing analysis report
==========================================

Report for net_6:
-----------------

Net not found: net_6
```

Yet, icetime seems to be aware of net_6's existence, judging from the netlist dump:

```
  ClkMux inmux_17_25_6_71816 (
    .I(net_6),
    .O(net_71816)
  );
  ClkMux inmux_21_24_6_88000 (
    .I(net_6),
    .O(net_88000)
  );
```

cliffordwolf, 2017-02-26:
I've now added `icetime -N` in commit 7d98cd8. It lists the nets allowed as argument to `-T`.

Afaict there are a two major misunderstandings here with respect to `-T`:

1. The "net" passed to the `-T` option is a physical wire on the chip, not a wire in the current design.

2. Icetime will print the longest path leading to that net. If the net is a clock, you will see the path driving that clock, not run timing analysis for the circuitry driven by the clock.

If you want to run more advanced timing analysis, such as multi-clock scenarios, use icetime to generate a timing netlist (`icetime -o`) and use a real timing analysis tool (such as OpenTimer) to run the actual analysis.

(It is on my todo list to test icetime + OpenTimer. So far I did not get to it.)

closed 2017-02-26 by cliffordwolf

madscientist159, 2017-02-26:
Makes sense.  Thanks for the timely update and clarification!