YosysHQ/icestorm #83: Inconsistency between icePLL and iCEcube2

C-Elegans, 2017-06-23:
When `icepll` is used to generate a PLL for an input frequency of 12MHz and an output of 25MHz, it gives this output:
```
F_PLLIN:    12.000 MHz (given)
F_PLLOUT:   25.000 MHz (requested)
F_PLLOUT:   24.000 MHz (achieved)

FEEDBACK: SIMPLE
F_PFD:   12.000 MHz
F_VCO:  768.000 MHz

DIVR:  0 (4'b0000)
DIVF: 63 (7'b0111111)
DIVQ:  5 (3'b101)

FILTER_RANGE: 1 (3'b001)
```
However, when iCEcube2 is given the same parameters, it generates a PLL with these parameters:
```
DivR: 0000 (0)
DivF: 1000010 (66)
DivQ: 101 (5)
Output frequency: 25.13MHz
```
It seems that iCEcube2 sets DIVF outside its documented range of [0,63], and gives a PLL configuration that more closely matches the frequency requested. Could the DIVF range given in the datasheet simply be a mistake? 

I can't verify this at the moment as I don't have an oscilloscope.

cliffordwolf, 2017-06-24:
> Could the DIVF range given in the datasheet simply be a mistake?

Good question. It would not be the first mistake in their datasheets.

I would guess that this is something that one could actually ask lattice support: "Your tool creates configs that are out of range according to your data sheet." No need to mention Project IceStorm at all..

C-Elegans, 2017-06-24:
Ok, I have experimentally verified that DIVF settings up to 88 do indeed work and produce the correct clock frequency (I generated a 440Hz tone and verified it with a tuner). I can't verify it for higher DIVF settings as my VCO frequency would be too high, but I suspect they will work too.I have also emailed lattice, but not heard back yet.

C-Elegans, 2017-06-24:
Interesting, the iCEcube software does seem to limit DIVF to 63 when using a feedback mode other than SIMPLE

C-Elegans, 2017-07-03:
Lattice replied, the PLL DIVF range is 0-127 (At least in the simple mode)

"Hi C_Elegans,

This should be an error in TN1251. Currently in ICE40 PLL module, the DIVF is a 7bit field in HW design(in picture blow). So the range should be from 0 to 127.

Seems TN1251 need to be updated. I will raise CR to get the document updated.
Regards"

cliffordwolf, 2017-07-03:
Thanks for sorting this out! I've now merged your PR that sets the DIVF range to [1, 127] in simple mode.

closed 2017-07-03 by cliffordwolf