YosysHQ/icestorm #86: Build system bikesheds

rqou, 2017-07-17:
I currently provide [automated nightly builds](https://rqou.com/jenkins/job/open-fpga-tools/) of the icestorm toolchain, and I am finally getting around to upstreaming the horrible ugly build system hacks that I've been using. I noticed that icestorm has a special `mxebin` makefile target that is used to create a Windows package. Is there a particular reason why the normal `install` target cannot be fixed to work? Would you accept a patch that fixes `install` for cross-building Windows packages?

cliffordwolf, 2017-07-17:
Usually when cross building something you don't want to install it (locally), and windows does not follow the usual LSB directory layout with bin directories and such. That's why `make mxebin` creates a .zip file with a "flat" directory structure.

rqou, 2017-07-17:
The way I've been doing it is to create an LSB-style directory layout anyways on Windows. I produce packages using `make DESTDIR=<foo>`. Since you've accepted my changes, I guess it is now possible to do it either way?

cliffordwolf, 2017-07-17:
>  I guess it is now possible to do it either way?

Yep. I think i can close this now? Just post if there is more to do here.

closed 2017-07-17 by cliffordwolf