YosysHQ/icestorm #95: High-level configuration tool license

rqou, 2017-08-27:
@rlutz:

At the risk of inciting a huge flamewar, I would like to request that the newly-added high-level configuration tool be licensed under a non-copyleft license rather than the copyleft GPLv3+. The rest of the IceStorm project is under the ISC license.

lowfatcomputing, 2017-08-27:
@rqou [ISC is compatible with GPL](https://www.gnu.org/licenses/license-list.en.html#ISC)

As an aside: Open fpga tools are something that would benefit from GPL in the future if the idea is to accrue a larger collection of supported devices. Linux benefits from accumulating device drivers because of GPL. The open source fpga tools are unlikely to get monentary support or future corporate contributions without any form of copyleft.

rlutz, 2017-08-27:
> As an aside: Open fpga tools are something that would benefit from gpl in the future if the idea is to accrue a larger collection of supported devices. Linux benefits from accumulating device [...]

@lowfatcomputing Thank you, I could not have said this in a better way.

cliffordwolf, 2017-08-27:
Oops. I did not notice the licence header. Thanks @rqou.

@rlutz: Either you put this under ISC or I will remove the files from the repository.

cliffordwolf, 2017-08-31:
No reaction in 4 days.

I've now removed the following files in commit 8354bc6 as they have GPL licenses:

- icebox/icebox_asc2hlc.py
- icebox/icebox_hlc2asc.py
- icebox/tc_logic_xpr.py
- icebox/tc_rxlat_netnames.py
- icebox/tc_xlat_netnames.py
- iceprog/iceprog.1

cc @martinda

closed 2017-08-31 by cliffordwolf