YosysHQ/icestorm #96: ParseError in high-level configuration tool

rqou, 2017-08-27:
cc @rlutz 

I just tried running icebox_asc2hlc.py and icebox_hlc2asc.py on a "real" design, and icebox_hlc2asc.py fails to round-trip the design. It instead raises a ParseError:

```
Traceback (most recent call last):
  File "./icebox/icebox_hlc2asc.py", line 1052, in <module>
    main()
  File "./icebox/icebox_hlc2asc.py", line 1049, in main
    main1(args[0])
  File "./icebox/icebox_hlc2asc.py", line 985, in main1
    stack[-1].read(fields)
  File "./icebox/icebox_hlc2asc.py", line 882, in read
    super().read(fields)
  File "./icebox/icebox_hlc2asc.py", line 754, in read
    raise ParseError
__main__.ParseError
```

The high-level bitstream file is [here](https://gist.github.com/rqou/c0b29322bb7bb13111c1039b3f3b03b7). Let me know if you would like me to try to minimize it.

rlutz, 2017-08-27:
Thank you for reporting this. I pushed a fix to https://github.com/rlutz/icestorm/tree/dev.

In order to help me understand a better high-level representation of PLLs, could you please provide me with a set of input files (`.v`, `.pcf`) and associated `.asc` output which contains just the PLL from this design (or another real-world example), with the used inputs/outputs connected to I/O pins?

rqou, 2017-08-27:
The .asc file is [here](https://gist.github.com/rqou/24637491c9d3af5484a93772f63fbaef), the .pcf is [here](https://github.com/rqou/laughing-waffle/blob/master/fpga/hx8k-breakout-pins.pcf), and the .v is [here](https://github.com/rqou/laughing-waffle/blob/master/fpga/cdb_dump.v) (warning: very hacky and low-quality Verilog; should not be used as an example). Hope this helps.

rlutz, 2017-08-27:
Thank you, I could create a minimal example from that. I will have to look into this in more detail tomorrow.