YosysHQ/icestorm #98: libftdi1-dev required on debian stretch

laf0rge, 2017-09-03:
When following the build instructions (copy+pasted the apt-get), it seems libftdi1-dev is missing:

```
make[1]: Entering directory '/space/home_laforge/projects/git/icestorm/iceprog'
cc -MD -O0 -ggdb -Wall -std=c99 -I/usr/local/include -I/usr/local/include   -c -o iceprog.o iceprog.c
iceprog.c: In function ‘main’:
iceprog.c:387:13: error: ‘INTERFACE_C’ undeclared (first use in this function); did you mean ‘INTERFACE_B’?
     ifnum = INTERFACE_C;
             ^~~~~~~~~~~
             INTERFACE_B
iceprog.c:387:13: note: each undeclared identifier is reported only once for each function it appears in
iceprog.c:389:13: error: ‘INTERFACE_D’ undeclared (first use in this function); did you mean ‘INTERFACE_C’?
     ifnum = INTERFACE_D;
             ^~~~~~~~~~~
             INTERFACE_C
iceprog.c:589:7: warning: implicit declaration of function ‘ftdi_usb_open_string’; did you mean ‘ftdi_usb_get_strings’? [-Wimplicit-function-declaration]
   if (ftdi_usb_open_string(&ftdic, devstr)) {
       ^~~~~~~~~~~~~~~~~~~~
       ftdi_usb_get_strings
<builtin>: recipe for target 'iceprog.o' failed
make[1]: *** [iceprog.o] Error 1
make[1]: Leaving directory '/space/home_laforge/projects/git/icestorm/iceprog'
Makefile:6: recipe for target 'all' failed
make: *** [all] Error 2
```

After installing the relted package, the build went through successful.

cliffordwolf, 2017-09-03:
Hi Harald, long time no see.

The problem you describe is odd. iceprog should work with both `libftdi-dev` and `libftdi1-dev`, and it does here on Ubuntu 16.04 LTS. (`libftdi-dev` is listed in the apt-get command line on the website.) For some reason it seems like on your system the ftdi.h from `libftdi-dev` does not have INTERFACE_C or INTERFACE_D.

But when I download `libftdi_0.20.orig.tar.gz` from https://packages.debian.org/stretch/libftdi-dev and open `src/ftdi.h` from that archive then I can see all four interfaces declared:

```
/** Port interface for chips with multiple interfaces */
enum ftdi_interface
{
    INTERFACE_ANY = 0,
    INTERFACE_A   = 1,
    INTERFACE_B   = 2,
    INTERFACE_C   = 3,
    INTERFACE_D   = 4
};
```

Is it possible that you have an older `ftdi.h` (for example in `/usr/local/include/`) that takes precedence of the one installed in `/usr/include`?

laf0rge, 2017-09-03:
My apologies, indeed there was an ancient /usr/local/include/fdti.h lingering around.

closed 2017-09-03 by laf0rge