YosysHQ/icestorm #102: `.comment' format mismatch

rlutz, 2017-09-04:
The syntax of the `.comment` statement expected by IceStorm does not appear to match the syntax output by `arachne-pnr`.

`icepack` expects comments to be in the following syntax:

    .comment
    First line
    Second line

Everything following `.comment` on the first line is ignored. Each subsequent line is treated as a comment until the comment block is terminated by a line starting with `.`. This line is treated normally; it is not ignored. If multiple comment blocks are specified, all except the last one are ignored.

`icepack` includes the comments in the following form at the beginning of the `.bin` output:

    ff 00 <First line> 00 <Second line> 00 00 ff

However, `arachne-pnr` outputs a comment of the following form:

    .comment arachne-pnr <version> (git sha1 <commit>, <compiler version and options>)

This is understood by `icepack` as an empty comment block, so it adds the pattern

    ff 00 00 ff

at the beginning of every `.bin` file generated from `arachne-pnr` output.

I stumbled across this when I realized that `icepack` actually interprets `.comment` statement. I'm not sure what the correct behavior is supposed to be, but I guess it's one of the following:

* `.comment` blocks are intended as a mechanism to include arbitrary strings in a `.bin` file, but `arachne-pnr` got the format wrong, so instead of the version string, an empty comment block is included in the output
* `.comment` blocks are intended as a mechanism to include arbitrary strings in a `.bin` file, but `arachne-pnr` mistook the syntax for a “real” comment and inadvertently inserts an empty comment block in every file
* `.comment` statements are intended as a way to be able to add comments to an `.asc` file; `icepack` is not supposed to include their content in the output

cliffordwolf, 2017-09-21:
As far as I am concerned this is a bug in arachne-pnr. @cseed ?

swillner, 2020-07-31:
Just confirming `nextpnr` also puts its comment on the same line
```
    out << ".comment from next-pnr" << std::endl;
```
https://github.com/YosysHQ/nextpnr/blob/master/ice40/bitstream.cc#L300

By the logic of the other statements, maybe this could be the expected behaviour?

byteit101, 2023-05-27:
I Just noticed this when trying to create 4-byte aligned output via comments. I can create a patch to support trailing/one line comments if people are ok with that?