YosysHQ/icestorm #105: [For FAQ?] UDEV Magic

drawkula, 2017-10-14:
From [Project Icestorm homepage](http://www.clifford.at/icestorm/):

> Notes for Linux: Create a file /etc/udev/rules.d/53-lattice-ftdi.rules with the following line in it to allow uploading bit-streams to a Lattice iCEstick and/or a Lattice iCE40-HX8K Breakout Board as unprivileged user:
>
> `ACTION=="add", ATTR{idVendor}=="0403", ATTR{idProduct}=="6010", MODE:="666"`

This did not work for me on Debian9. Other tries like setting the device node's ownership/permissions right for my account failed too.

After digging a while, I found Debian's `fpga-icestorm_*.deb` includes this UDEV rule:

    ATTRS{idVendor}=="0403", ATTRS{idProduct}=="6010", MODE="0660", GROUP="plugdev", TAG+="uaccess"

This one worked here and now I can drop `sudo` for loading the FPGA.

dh73, 2017-10-14:
The good Debian.
I can comment this: From the Security Implications of System Groups:

**To manage removable drives without root privileges, you have to use the groups cdrom** for optical drives. **Optical drives can usually be mounted in according to the rules in /etc/fstab, but for other removable drives the group plugdev may be more practical**. 

And as in the both udev rules, the last one contains the group membership correctly, so for this reason you're able to use udev/pmount without root privileges. 

Something you can check is what groups your user belongs to, and if you add yourself to plugdev (I'll probably will not do that if this is a multi user - critical server) you should be able to stick in the original udev rule.

Source:
https://wiki.debian.org/SystemGroups

drawkula, 2017-10-14:
I could not use the 1st mentioned UDEV rule desoite being in the right group.

`TAG+="uaccess"` seems to be what makes the difference...

rqou, 2017-11-15:
Heh, azonenberg/openfpga had this same issue. According to @cyrozap, 
<blockquote>`TAG+="uaccess"` is for systemd compatibility and `MODE="660", GROUP="plugdev"` is for non-systemd systems. This is the same pattern OpenOCD uses for its udev rules.</blockquote>

So the proposed change to the FAQ should be correct.

drawkula, 2017-11-15:
Ahhhh... systemd stuff... good to have an explanation!
Thanks!

cliffordwolf, 2017-11-18:
Updated the website.

closed 2017-11-18 by cliffordwolf

eddyb, 2019-12-07:
I feel kinda dumb after actually reading the line, as it clearly says `GROUP="plugdev"`, but maybe it would be helpful to explicitly say next to it that you *also* need to be in the `plugdev` group for this to apply?