YosysHQ/icestorm #111: Upduino ICE40UP5k ICEPROG doesnt  work.

j0t4, 2017-12-04:
I tried with rgb example here ...
https://github.com/cliffordwolf/icestorm/tree/master/examples/up5k_rgb

I used the same connections as a working setup with Lattice Diamond programmer, FTDI2232H (channel A). 

I get no errors on "sudo make prog" after successful "make",  but fpga  don't get  programmed.
I can see ft2232 pins "talk" to fpga board when i type "sudo make prog" with a logic analyzer but  I have no experience with SPI or  jtag to go further.
 
Someone has a working setup with RGB example trough "Makefile" and "make prog" con FTDI2232?
THanks

daveshah1, 2017-12-04:
I suspect the problem is that the example "make prog" is set up to program SRAM, not flash. Either remove the flash CS jumper and swap MISO and MOSI to program SRAM, or change the target Makefile from
```
$(ICEPROG) -S $<
```
to
```
$(ICEPROG) $<
```

j0t4, 2017-12-05:
Thanks !  
Working like a charm!

closed 2017-12-05 by j0t4

mkvenkit, 2019-04-22:
@daveshah1  Thanks for this - just saved me some time!