YosysHQ/icestorm #114: icetime - 1000020.4ns

igor-m, 2018-01-12:
Does the 10000XX.XX indicate a design issue here (latest pull from github):
`icetime topological timing analysis report
==========================================

Report for critical path:
-------------------------

        loop-start at net_34778
        t7262 (LocalMux) I -> O: 0.330 ns
        inmux_9_15_38650_38720 (InMux) I -> O: 0.260 ns
        t1267 (CascadeMux) I -> O: 0.000 ns
        lc40_9_15_7 (LogicCell40) in2 -> lcout: 0.379 ns
1000000.968 ns net_34778 (_j1.insn_from_memory[13])
        odrv_9_15_34778_38506 (Odrv4) I -> O: 0.372 ns
        t7288 (LocalMux) I -> O: 0.330 ns
        inmux_10_14_42377_42392 (InMux) I -> O: 0.260 ns
        t1388 (CascadeMux) I -> O: 0.000 ns
        lc40_10_14_1 (LogicCell40) in2 -> lcout: 0.379 ns
1000002.307 ns net_38480 ($abc$26981$n3234_1)
        odrv_10_14_38480_42090 (Odrv4) I -> O: 0.372 ns
        t8348 (Span4Mux_h4) I -> O: 0.316 ns
        t8347 (Span4Mux_h4) I -> O: 0.316 ns
        t8346 (Span4Mux_h1) I -> O: 0.175 ns
        t8345 (LocalMux) I -> O: 0.330 ns
        inmux_20_14_80032_80082 (InMux) I -> O: 0.260 ns
        lc40_20_14_3 (LogicCell40) in3 -> lcout: 0.316 ns
1000004.390 ns net_76486 ($abc$26981$n3939_1)
        odrv_20_14_76486_53935 (Odrv12) I -> O: 0.540 ns
        t15878 (Span12Mux_h2) I -> O: 0.168 ns
        t15877 (LocalMux) I -> O: 0.330 ns
        inmux_11_14_46195_46251 (InMux) I -> O: 0.260 ns
        lc40_11_14_6 (LogicCell40) in0 -> lcout: 0.449 ns
1000006.137 ns net_42316 ($abc$26981$n3938_1)
        t8921 (LocalMux) I -> O: 0.330 ns
        inmux_11_13_46065_46106 (InMux) I -> O: 0.260 ns
        t1542 (CascadeMux) I -> O: 0.000 ns
        lc40_11_13_2 (LogicCell40) in2 -> lcout: 0.379 ns
1000007.105 ns net_42189 ($abc$26981$n3937_1)
        odrv_11_13_42189_45906 (Odrv12) I -> O: 0.540 ns
        t8884 (Span12Mux_h4) I -> O: 0.217 ns
        t8883 (Sp12to4) I -> O: 0.449 ns
        t8882 (Span4Mux_v0) I -> O: 0.203 ns
        t8881 (Span4Mux_h0) I -> O: 0.147 ns
        t8880 (Span4Mux_v1) I -> O: 0.203 ns
        t8879 (LocalMux) I -> O: 0.330 ns
        inmux_9_12_38294_38319 (InMux) I -> O: 0.260 ns
        lc40_9_12_2 (LogicCell40) in0 -> lcout: 0.449 ns
1000009.903 ns net_34404 (_j1.pcN[1])
        odrv_9_12_34404_34311 (Odrv4) I -> O: 0.372 ns
        t7065 (Span4Mux_v4) I -> O: 0.372 ns
        t7064 (Span4Mux_h4) I -> O: 0.316 ns
        t7063 (Span4Mux_v4) I -> O: 0.372 ns
        t7082 (Span4Mux_h4) I -> O: 0.316 ns
        t7081 (Span4Mux_h4) I -> O: 0.316 ns
        t7080 (Span4Mux_h4) I -> O: 0.316 ns
        t7108 (Span4Mux_h0) I -> O: 0.147 ns
        t7107 (Span4Mux_v4) I -> O: 0.372 ns
        t7106 (Span4Mux_h0) I -> O: 0.147 ns
        t7105 (Span4Mux_v4) I -> O: 0.372 ns
        t7104 (Span4Mux_h0) I -> O: 0.147 ns
        t7103 (Span4Mux_v4) I -> O: 0.372 ns
        t7102 (Span4Mux_h4) I -> O: 0.316 ns
        t7101 (Span4Mux_v4) I -> O: 0.372 ns
        t7100 (Span4Mux_v4) I -> O: 0.372 ns
        t7099 (Span4Mux_v4) I -> O: 0.372 ns
        t7098 (Span4Mux_h4) I -> O: 0.316 ns
        t7097 (Span4Mux_v4) I -> O: 0.372 ns
        t7096 (Span4Mux_v2) I -> O: 0.252 ns
        t7095 (IoSpan4Mux) I -> O: 0.323 ns
        t7094 (Span4Mux_h0) I -> O: 0.147 ns
        t7093 (Span4Mux_v4) I -> O: 0.372 ns
        t7092 (Span4Mux_v4) I -> O: 0.372 ns
        t7091 (Span4Mux_v4) I -> O: 0.372 ns
        t7090 (Span4Mux_h0) I -> O: 0.147 ns
        t7089 (Span4Mux_v4) I -> O: 0.372 ns
        t7088 (Span4Mux_h0) I -> O: 0.147 ns
        t7087 (Span4Mux_v4) I -> O: 0.372 ns
        t7086 (Span4Mux_h0) I -> O: 0.147 ns
        t7085 (Span4Mux_v4) I -> O: 0.372 ns
        t7084 (Span4Mux_h2) I -> O: 0.203 ns
        t7083 (LocalMux) I -> O: 0.330 ns
        inmux_19_3_75447_75480 (InMux) I -> O: 0.260 ns
        t2677 (CascadeMux) I -> O: 0.000 ns
1000020.143 ns net_75480_cascademuxed
        ram_19_3 (SB_RAM40_4K) RADDR[1] [setup]: 0.290 ns
1000020.433 ns dangling_wire_513

Resolvable net names on path:
1000000.968 ns ..1000001.929 ns _j1.insn_from_memory[13]
1000002.307 ns ..1000004.075 ns $abc$26981$n3234_1
1000004.390 ns ..1000005.688 ns $abc$26981$n3939_1
1000006.137 ns ..1000006.726 ns $abc$26981$n3938_1
1000007.105 ns ..1000009.454 ns $abc$26981$n3937_1
1000009.903 ns ..1000020.143 ns _j1.pcN[1]
              RDATA[11] -> _bn20.rd[11]
               RDATA[3] -> _bn20.rd[3]

Total number of logic levels: 7
Total path delay: 1000020.43 ns (0.00 MHz)`

daveshah1, 2018-01-18:
The problem is an inferred latch in your ram_test.v, because the decoder doesn't cover all possibilities. Without totally rewriting the decoder, the simplest fix I see is to add an "else" clause as follows (choose your own default value) :
```
    always @*
    begin
      casez ({c12, c11}) // Depending on memory address, select different RAM blocks.
      2'b00: insn = insn0;
      2'b01: insn = insn1;
      2'b10: insn = insn2;
      2'b11: begin
			if (!c10 & !c9 & !c8) insn = insn3;   // Select the 256x16 blocks
			else if (!c10 & !c9 &  c8) insn = insn4;
			else if (!c10 &  c9 & !c8) insn = insn5;
			else if (!c10 &  c9 &  c8) insn = insn6;
			else if ( c10 & !c9 & !c8) insn = insn7;
			else if ( c10 & !c9 &  c8) insn = insn8;
			else insn = 16'h0000;
			end
	  
      endcase
```
With the fix you get a timing estimate of 37.25MHz.

daveshah1, 2018-01-18:
I will keep the issue open however, as a reminder to myself to add a nicer error message in icetime in this situation.

cliffordwolf, 2018-01-18:
> the simplest fix I see is to add an "else" clause as follows (choose your own default value)

I'd personally prefer a default assignment in the beginning of the block. something like this:

```
    always @*
    begin
      insn = 16'bx;
      casez ({c12, c11}) // Depending on memory address, select different RAM blocks.
      2'b00: insn = insn0;
      2'b01: insn = insn1;
      2'b10: insn = insn2;
      2'b11: begin
			if (!c10 & !c9 & !c8) insn = insn3;   // Select the 256x16 blocks
			else if (!c10 & !c9 &  c8) insn = insn4;
			else if (!c10 &  c9 & !c8) insn = insn5;
			else if (!c10 &  c9 &  c8) insn = insn6;
			else if ( c10 & !c9 & !c8) insn = insn7;
			else if ( c10 & !c9 &  c8) insn = insn8;
			end
      endcase
    end
```

igor-m, 2018-01-18:
OMG, a freshman's mistake :) Thanks!
The timing estimates from 32-40MHz based on the seed..

Zaba, 2022-10-13:
I get a similar report from icetime (with loop-start, the time starting from ~1000000 ns, etc.) with the following Verilog module:

```verilog
module repro(clk, out);

input		clk;
output		out;
reg	[8:0]	data;

always @(posedge clk) begin
	if (data == 9'd 10)
		data <= 9'b 0;
	else
		data <= data + 1;
end

assign out = data < 9'd 8;

endmodule
```

Changing the register width from `[8:0]` to `[7:0]` makes the icetime report look normal. The issue seems to originate from the `data < 9'd 8` comparison. I don't see any reports in the Yosys log of any inferred latches or any other abnormalities.

Putting aside the question of whether this is good Verilog, I'd really like to figure out why the icetime report looks the way it does.

Yosys/Nextpnr/icetime output: [yosys-nextpnr-icetime.log](https://github.com/YosysHQ/icestorm/files/9780022/yosys-nextpnr-icetime.log)