YosysHQ/icestorm #115: UP5k Arachne dump

igor-m, 2018-01-12:
While building this (yesterday's pull and reinstall of the icestorm/arachne/yosys)
https://github.com/igor-m/UPduino-Mecrisp-Ice-15kB/tree/master/icestorm%20version
I get an arachne-pnr dump

```
  After placement:
PIOs       21 / 39
PLBs       541 / 660
BRAMs      30 / 30

  place time 22.68s
route...
36 -> 77837 75389 76613 26219 77633 75593 28667 28463 78041 76205 77021 27647 77225 75797 28871 28055 78241 76409 76817 27851 77429 76001 29071 28259 26423 26627 26831 27035 27239 27443 78429
arachne-pnr: src/route.cc:793: void Router::route(): Assertion `unrouted.empty()' failed.
Aborted (core dumped)
```

Not sure it comes from the design itself or from the experimental UP5k..

daveshah1, 2018-01-13:
This now seems to be fixed in my branch [up5k_ip2](https://github.com/daveshah1/icestorm/tree/up5k_ip2), commit f91b2a8. I will submit a pull request with this fix as well as several other UltraPlus related fixes and improvements in the next couple of days.

igor-m, 2018-01-15:
Will try with final, thanks.

igor-m, 2018-01-18:
I've done in my icestorm dir (ubuntu)
git pull
make -j$(nproc)
sudo make install
in order to include the recent UP5k fixes, but still getting the arachne dump.

closed 2018-01-18 by igor-m

igor-m, 2018-02-18:
With the latest update of the tools and with additions in j1a.v I get the dump again:
`Synthesize design...
Warning: Resizing cell port $paramod$6eb35da5f2126d454a6400e52ebfca14e8128678\SB_RAM256x16._ram1.WADDR from 8 bits to 11 bits.
Warning: Resizing cell port $paramod$6eb35da5f2126d454a6400e52ebfca14e8128678\SB_RAM256x16._ram1.RADDR from 8 bits to 11 bits.
Warning: Resizing cell port top.ramfn_inst3.ADDRESS from 16 bits to 14 bits.
Warning: Resizing cell port top.ramfn_inst2.ADDRESS from 16 bits to 14 bits.
Warning: Resizing cell port top.ramfn_inst1.ADDRESS from 16 bits to 14 bits.
Warning: Resizing cell port top.ramfn_inst0.ADDRESS from 16 bits to 14 bits.
Warning: Resizing cell port top._bn35.WADDR from 11 bits to 8 bits.
Warning: Resizing cell port top._bn35.RADDR from 11 bits to 8 bits.
Warning: Resizing cell port top._bn34.WADDR from 11 bits to 8 bits.
Warning: Resizing cell port top._bn34.RADDR from 11 bits to 8 bits.
Warning: Resizing cell port top._bn33.WADDR from 11 bits to 8 bits.
Warning: Resizing cell port top._bn33.RADDR from 11 bits to 8 bits.
Warning: Resizing cell port top._bn32.WADDR from 11 bits to 8 bits.
Warning: Resizing cell port top._bn32.RADDR from 11 bits to 8 bits.
Warning: Resizing cell port top._bn31.WADDR from 11 bits to 8 bits.
Warning: Resizing cell port top._bn31.RADDR from 11 bits to 8 bits.
Warning: Resizing cell port top._bn30.WADDR from 11 bits to 8 bits.
Warning: Resizing cell port top._bn30.RADDR from 11 bits to 8 bits.
Place and Route...
seed: 3930196931
device: 5k
read_chipdb +/share/arachne-pnr/chipdb-5k.bin...
  supported packages: sg48, uwg30
read_blif j1a.blif...
prune...
read_pcf j1a.pcf...
instantiate_io...
pack...

After packing:
IOs          28 / 39
  IO_I3Cs    0 / 0
  IO_ODs     0 / 3
GBs          0 / 8
  GB_IOs     0 / 8
LCs          3346 / 5280
  DFF        1430
  CARRY      171
  CARRY, DFF 51
  DFF PASS   279
  CARRY PASS 10
BRAMs        30 / 30
WARMBOOTs    0 / 1
PLLs         0 / 1
MAC16s       0 / 8
SPRAM256KAs  4 / 4
HFOSCs       0 / 1
LFOSCs       0 / 1
RGBA_DRVs    1 / 1
LEDDA_IPs    0 / 1
I2Cs         0 / 2
SPIs         0 / 2

place_constraints...
terminate called after throwing an instance of 'std::out_of_range'
  what():  map::at
Aborted (core dumped)
Create Bitstream...
Timing Analysis...
// Reading input .asc file..
// Reading 5k chipdb file..
// Creating timing netlist..
Warning: timing analysis not supported for cell type RGBA_DRV
Missing entry in inports_map for cell type SB_RGBA_DRV!
Finished!`

daveshah1, 2018-02-18:
The fact that it's showing 0 I3C IOs available suggests that you updated arachne-pnr without also updating icestorm. Can you make sure you have the latest icestorm installed and then rebuild and reinstall arachne-pnr?

I also see a separate issue where icetime correctly it doesn't support RGBA_DRV, but then kills the timing analysis instead of ignoring it. That's a separate issue that I'll sort out tomorrow.

igor-m, 2018-02-18:
I did yosys->arachne->icestorm update.. Will try w/ arachne again..

igor-m, 2018-02-18:
I've changed the IO assignments (pushed to github as well) similar to IceCueb and now it routes:
```
After placement:
PIOs       21 / 39
PLBs       600 / 660
BRAMs      30 / 30

  place time 44.77s
route...
  pass 1, 59 shared.
  pass 2, 22 shared.
  pass 3, 10 shared.
  pass 4, 8 shared.
  pass 5, 3 shared.
  pass 6, 3 shared.
  pass 7, 1 shared.
  pass 8, 0 shared.

After routing:
span_4     5879 / 23008
span_12    1073 / 4464

  route time 19.00s
write_txt j1a.txt...
Create Bitstream...
Timing Analysis...
// Reading input .asc file..
// Reading 5k chipdb file..
// Creating timing netlist..
Warning: timing analysis not supported for cell type RGBA_DRV
Missing entry in inports_map for cell type SB_RGBA_DRV!
Finished!
```

igor-m, 2018-02-19:
Added your fix and get
```
Timing Analysis...
// Reading input .asc file..
// Reading 5k chipdb file..
// Creating timing netlist..
Warning: timing analysis not supported for cell type HFOSC
Warning: timing analysis not supported for cell type RGBA_DRV
// Timing estimate: 56.88 ns (17.58 MHz)
// Checking 27.78 ns (36.00 MHz) clock constraint: FAILED.
Finished!
```
Interestingly it works fine at 24MHz (with MAC16 16x16 multiplier).

daveshah1, 2018-02-19:
That sounds about right. Timing analysis is conservative for process variations, voltage and temperature. So at room temperature, a decent device and stable 1.2V Vcc, it's not surprising that a design giving an estimate of 17MHz runs at 24MHz - but it shouldn't be reliabled upon for production use of course.