YosysHQ/icestorm #117: UP5k - icestorm/icetime - size and time estimates

igor-m, 2018-01-20:
When running this https://github.com/igor-m/UPduino-Mecrisp-Ice-15kB/tree/master/icestorm%20version with icestorm (the latest) and IceCube2 I get two quite different results:
1. IceCube2
size 383 PLBs, timing estimate 31.5MHz, works fine at 20/30MHz external osc.
2. icestorm
size 541-557 PLBs, timing estimate 32-40MHz, does not run reliably at 20MHz ext. osc.

daveshah1, 2018-01-20:
The difference in size is not that surprising, arachne-pnr's packing is definitely suboptimal. The biggest worry is it not working reliably. Does this depend on the seed at all? Does it work reliably at half the clock?

igor-m, 2018-01-20:
The way how it corrupts incoming data under icestorm/20MHz is typical for IceCube version when overclocked. It seems to me the icetime timing estimate does not show proper numbers in this case, because the icestorm uses much more PLBs the design is actually much slower. I will try with slower clock.

igor-m, 2018-01-20:
With 14.318MHz the same - typically when holding "Enter" it corrupts the "ok."
  ok.
  ok.
  ok.
  ok.
 ?
  ok.
  ok.
  ok.
  ok.
  ok.
  ok.
þ ?
þ ?
  ok.
  ok.
  ok.
  ok.
ÿ ?
  ok.

daveshah1, 2018-01-20:
Interesting. Can you see what happens with a couple of different arachne-pnr seeds?

igor-m, 2018-01-20:
The same at 7.2MHz clock.

cliffordwolf, 2018-01-20:
Your design is broken: RXD isn't synchronized to the system clock. I'm pretty sure your problems will go away if you properly synchronize your asynchronous inputs.

igor-m, 2018-01-20:
I can add a synchronizer at the RXD, sure. But with current design I've done 12x runs with icestorm (different seeds) all with errors, 4x icecube no errors (20MHz, the same HW and the same verilog).

igor-m, 2018-01-20:
Added 3FF synchroniser (j1a.v)
`  // ######   UART   ##########################################

  // an RXD input synchroniser

  reg rxd1, rxd2, rxd3;
  always @(posedge clk) 
	begin
	rxd1 <= RXD;
	rxd2 <= rxd1;
	rxd3 <= rxd2;
	end  

  wire uart0_valid, uart0_busy;
  wire [7:0] uart0_data;
  wire uart0_wr = io_wr & (mem_addr == `adr_uart0);
  wire uart0_rd = io_rd & (mem_addr == `adr_uart0);
  wire UART0_RX;
  buart _uart0 (
     .clk(clk),
     .resetq(1'b1),
     .rx(rxd3),`
No errors so far..

daveshah1, 2018-01-20:
Good, I think this can safely be closed then :) The issues with PLB packing efficiency will improve when/if VPR is ported to the iCE40.

closed 2018-01-20 by cliffordwolf

igor-m, 2018-01-20:
With the RXD synchronizer, icestorm:
1. works fine at 20MHz external xtal osc.
2. does not work at 30MHz ext. xtal osc. at all (a crash), timing estimates 36-38MHz.
I will close this, and will try with different uarts later on.
Thanks!

daveshah1, 2018-01-20:
Ok, it seems icetime is wildly optimistic (in the worst case a factor of two out) compared to icecube's timing analysis. 

The problem seems to be that there is duplicate data in the timing database, with many cell types having multiple different values, and in some cases the lower rather than higher delay ends up being used, whereas the higher delay must be accurate. The reason for these duplicates in the first place is a mystery though.

daveshah1, 2018-01-20:
Bad timing data is removed in 78e5b89. Results of icetime are much closer now.

igor-m, 2018-01-21:
With 19-22MHz the timing estimates here match what I see in situ much better..

igor-m, 2018-01-21:
Size: could it be the excessive ~150 PLBs come from an 16x16multiplier in the icestorm version?
The IceCube infers a DSP module for the j1.v line 57: "wire [31:0] umstar = st0 * st1;"

daveshah1, 2018-01-21:
Yes, that will be the problem. Yosys does not support DSP inference yet.