YosysHQ/icestorm #120: [glibc 2.26] Can't compile icestorm because of missing xlocale header.

C47D, 2018-02-02:
Hi,

Recently i updated my OS from Ubuntu 16.04 to 17.10 and tried to compile icestorm (I'm on the current master HEAD (722790a)).

When i try to compile icestorm i got:
```
icestorm $ make
for dir in icebox icepack iceprog icemulti icepll icetime icebram; do \
	make -C $dir all || exit; \
done
make[1]: Entering directory '/home/d3bug/icestorm/icebox'
make[1]: Nothing to be done for 'all'.
make[1]: Leaving directory '/home/d3bug/icestorm/icebox'
make[1]: Entering directory '/home/d3bug/icestorm/icepack'
make[1]: *** No rule to make target '/usr/include/xlocale.h', needed by 'icepack.o'.  Stop.
make[1]: Leaving directory '/home/d3bug/icestorm/icepack'
Makefile:6: recipe for target 'all' failed
make: *** [all] Error 2
```

So searching a bit i found [this](https://sourceware.org/glibc/wiki/Release/2.26#Removal_of_.27xlocale.h.27) release notes for glibc 2.26.
> The nonstandard header xlocale.h has been removed in this release.

To know the version of glibc on my box i used:
```
$ ldd --version      
ldd (Ubuntu GLIBC 2.26-0ubuntu2.1) 2.26
Copyright (C) 2017 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
Written by Roland McGrath and Ulrich Drepper.
```

So that why i can't compile icestorm nor arachne (i got the same error output). I tried searching xlocale on the icepack source but i couldn't find it (i haven't searched deeply tho).

I know this is not directly a icestorm problem but i wanted to ask for suggestions to solve it, i have found [this](https://www.linuxquestions.org/questions/slackware-14/no-luck-with-libc-on-current-xlocale-h-missing-4175612648/#post5752136) workaround tho.

Can this be solved directly on icestorm/arachne or should i find a way to downgrade glibc?

Regards.

daveshah1, 2018-02-03:
Are you on a clean copy of the repository? In particular, make sure you delete icepack.d and icepack.o and try again.

cliffordwolf, 2018-02-03:
What @daveshah1 said. You have old `.d` files in your repository. Run `make clean` or build from a fresh working copy.

closed 2018-02-03 by cliffordwolf

C47D, 2018-02-04:
Thanks, ```make clean``` and then ```make``` solved it.