YosysHQ/icestorm #121: iceprog sram programming does not work with icestick

ekordas, 2018-02-03:
I got a new Icestick to start experimenting with verilog. The normal programming works fine, but programming it with
`iceprog -S file.bin`
does not work. The leds on the stick then go into the half-on state and nothing else happens. Also it prints `cdone: low` at the end.
Attached is a wireshark usb capture, if that helps.
[program_iceprog_sram.zip](https://github.com/cliffordwolf/icestorm/files/1691686/program_iceprog_sram.zip)

daveshah1, 2018-02-03:
I don't think SRAM programming is possible on the icestick, without hardware modifications (swapping MOSI and MISO, and removing the CS connection the flash chip). You will have to use flash programming only otherwise.

cliffordwolf, 2018-02-03:
Quoting from `iceprog --help` output:

```
Notes for iCEstick (iCE40HX-1k devel board):
  An unmodified iCEstick can only be programmed via the serial flash.
  Direct programming of the SRAM is not supported. For direct SRAM
  programming the flash chip and one zero ohm resistor must be desoldered
  and the FT2232H SI pin must be connected to the iCE SPI_SI pin, as shown
  in this picture:
  http://www.clifford.at/gallery/2014-elektronik/IMG_20141115_183838
```

closed 2018-02-03 by cliffordwolf