YosysHQ/icestorm #125: icebox_vlog.py does not respect $(PREFIX) for chipdb location

thoughtpolice, 2018-02-14:
In `./icebox/icebox_vlog.py`:

https://github.com/cliffordwolf/icestorm/blob/88eebff7db9d76e418ccbddd884c4e617596a428/icebox/icebox_vlog.py#L378

This doesn't respect `$(PREFIX)` during installation, meaning `icebox_vlog` can't be used (for example, when fiddling with `picorv32`.) This is a problem for any package that wants to install Icestorm outside of `/usr/local`; [NixOS](https://nixos.org) is the one I'm using.

I'm not sure how this should be fixed; perhaps simply as a `sed` invocation in the `install:` target?

thoughtpolice, 2018-02-14:
Here is the `nixpkgs` fix: https://github.com/NixOS/nixpkgs/commit/fa8f4b187f703950a9e3119549d264ff7e233123

```diff
From fa8f4b187f703950a9e3119549d264ff7e233123 Mon Sep 17 00:00:00 2001
From: Austin Seipp <aseipp@pobox.com>
Date: Wed, 14 Feb 2018 16:09:42 -0600
Subject: [PATCH] icestorm: fix chipdb location for icebox_vlog

---
 pkgs/development/tools/icestorm/default.nix | 7 +++++++
 1 file changed, 7 insertions(+)

diff --git a/pkgs/development/tools/icestorm/default.nix b/pkgs/development/tools/icestorm/default.nix
index 6351a01501d3..0307027faa2e 100644
--- a/pkgs/development/tools/icestorm/default.nix
+++ b/pkgs/development/tools/icestorm/default.nix
@@ -15,6 +15,13 @@ stdenv.mkDerivation rec {
   buildInputs = [ python3 libftdi ];
   makeFlags = [ "PREFIX=$(out)" ];
 
+  # fix icebox_vlog chipdb path. icestorm issue:
+  #   https://github.com/cliffordwolf/icestorm/issues/125
+  patchPhase = ''
+    substituteInPlace ./icebox/icebox_vlog.py \
+      --replace /usr/local/share "$out/share"
+  '';
+
   meta = {
     description = "Documentation and tools for Lattice iCE40 FPGAs";
     longDescription = ''
```

daveshah1, 2018-02-16:
The other option would be to do the same as how arachne-pnr finds the chipdb, and use a path relative to the script (i.e. ../share/icebox/chipdb-*.txt)