YosysHQ/icestorm #127: icetime: invalid chipdb path on Windows

drizzd, 2018-02-21:
I cross-compile icetime for Windows with mingw-w64 using Windows Subsystem for Linux (essentially on Linux). So, it should be close to this https://github.com/FPGAwars/toolchain-icestorm/tree/master/scripts. I set PREFIX to `/mnt/c/Users/Ricl/pkgs/icestorm-8e84261`. I am not using Msys or Cygwin, which would translate the UNIX path to a Windows path. Therefore, icetime will try to fopen `/mnt/c/Users/Ricl/pkgs/icestorm-8e84261/share/icebox/chipdb-1k.txt`, which is not a valid path for msvcrt.dll.

```
$ icetime -tmd hx1k rot.txt
// Reading input .asc file..
// Reading 1k chipdb file..
Can't open chipdb file: No such file or directory
```

It works fine if I use the `-C` option:

```
$ icetime -C 'C:/Users/ricl/pkgs/icestorm-8e84261/share/icebox/chipdb-1k.txt' -tmd hx1k rot.txt
// Reading input .asc file..
// Reading 1k chipdb file..
// Creating timing netlist..
[... snip ...]
```

How about we instead determine the chipdb path dynamically, relative to the icetime executable path. Arachne-pnr provides a function `proc_self_dirname` for this purpose. Otherwise I don't see how we can properly do this when cross-compiling.

benpye, 2019-03-17:
@drizzd Wondering if you ever pursued this? Looking to get yosys + icestorm + nextpnr working on Windows for myself. Might try and set it up in CI if I can get it going.

drizzd, 2019-03-31:
I think I was just using `-C`. I did not find any code. I avoid writing code when there is no prior contact with the maintainer.

daveshah1, 2019-03-31:
Looks like support for this was added in #184. Maybe you could test again and see if it is working for you?

drizzd, 2019-03-31:
Thanks. It looks like that should resolve this issue. @benpye can you test? Currently I am to build libftdi and so I cannot build icestorm either.

closed 2019-04-14 by drizzd