YosysHQ/icestorm #134: 5k LED driver pin mapping now broken

JanCoombs, 2018-04-21:
When I saw iCE40 5K part support I got a great board from http://gnarlygrey.atspace.cc/development-platform.html     The demo code was easily adapted, and the RGB LED demo worked. 

Please add these two simple 5k boards to your list. 

A few weeks later the code failed with:
arachne-pnr -d 5k -p RGB_LED_BLINK.pcf RGB_LED_BLINK.blif -o RGB_LED_BLINK.txt
seed: 1
device: 5k
...
place_constraints...
fatal error: bad constraint on `REDn': pin 41 does not correspond to RGB driver output `RGB0'
Makefile:3: recipe for target 'RGB_LED_BLINK.bin' failed
make: *** [RGB_LED_BLINK.bin] Error 1

I'm not sure where in the chain this error started, so have posted it here. 

Jan Coombs   jan4icestrm@murray-microft.co.uk

[RGB_LED_BLINK_2018-04-21_clean.tar.gz](https://github.com/cliffordwolf/icestorm/files/1934576/RGB_LED_BLINK_2018-04-21_clean.tar.gz)

daveshah1, 2018-04-21:
Thanks for the report! 

Unfortunately the reality is it didn't really work before, just the error checking was broken. Now at least it fails with an error. This is due to a difference in how the RGB primitive is implemented. 

arachne-pnr requires that RGB driver RGB0 connects to hardware RGB0. Before this was always happening, but not being enforced in the design, so the colours would be wrong (so RGB0 would always connect to pin 39 regardless of the design/pcf). icecube2 will swap the inputs correspondingly so that the RGB pins can be correctly swapped. But arachne-pnr can't do this automatically yet, so you will need to manually swap the inputs and outputs of the RGB driver so they line up with the hardware.

This is on my to-do list to fix, but I can't give a time estimate for when this will be fixed.

JanCoombs, 2018-04-21:
Great, thanks. I've twiddled the source so that it builds, and the wire labels are correct for the $15.99 postpaid Gnarly Grey UPDuino v2.0 Board  (http://gnarlygrey.atspace.cc/development-platform.html)

[RGB_LED_BLINK_2018-04-21_1544_clean.tar.gz](https://github.com/cliffordwolf/icestorm/files/1934658/RGB_LED_BLINK_2018-04-21_1544_clean.tar.gz)  1.3kB

closed 2018-04-21 by JanCoombs