YosysHQ/icestorm #137: Number of LUT's / BRAM's

acid-maker, 2018-05-09:
I use icestorm tools to program FPGA iCE40HX4K-TQ144.

According to specification this chip should have 3520 LUT's and 20 BRAM's, but when I invoke arachne-pnr (`arachne-pnr -d 8k -P tq144:4k ....`) I see something like this:

```
After packing:
IOs          76 / 107
GBs          0 / 8
  GB_IOs     0 / 8
LCs          2771 / 7680
  DFF        815
  CARRY      632
  CARRY, DFF 303
  DFF PASS   193
  CARRY PASS 111
BRAMs        17 / 32
WARMBOOTs    0 / 1
PLLs         1 / 2

```

In my opinion number of LUT's should be 3520 not 7680, also number of BRAMs should be 20 not 32.

But there is more - recently I made project that uses 24 BRAM's in my FPGA which supposed to have 20 BRAM's - because arachne thinks that I have 32 BRAM's it made bitstream. But the funny thing is that it actually works !!! Now I'm not sure if this is really issue with icestorm (probably arachne-pnr) or there are more BRAMs than there supposed to be in my FPGA.

daveshah1, 2018-05-09:
Hi @acid-maker!

There is no bug here - the 4k and 8k parts really are the same die inside, so you can use all 7680 LUTs and 32 BRAMs in the "4k" part with icestorm. The only difference to the 8k parts is the pinout, which is why arachne-pnr is called with "-d 8k" to tell it the chip inside is an 8k, but "-P tq144:4k", which means use pinout information for the "4k" device in TQFP144 package (in this case there is no TQFP144 8k anyway).

acid-maker, 2018-05-10:
Wow. This is exactly what I suspected. Now I'm little confused. Does anybody knows why they sell 8k chips as 4k chips? Are they fully functional or they test structures in production and faulty ones (with for example one faulty LUT) put in '4k' packages and fully functional in '8k' packages?

The question is - Can I safely use '4k' chips as '8k'?

daveshah1, 2018-05-10:
No-one has ever found a '4k' part with any faulty '8k' functionality, so you should be safe. 

It is plausible, as the '4k' package has fewer PLLs broken out that one of the '8k' PLLs is defective in the '4k' parts - but you would never notice that, and it would never be a problem, because you can't use them anyway because VccPLL and GNDPLL aren't broken out. But I suspect even this isn't the case.

closed 2018-06-15 by cliffordwolf