YosysHQ/icestorm #140: Ubuntu packages for the complete 'icestorm flow' ?

kehribar, 2018-05-24:
Hi all,

I, personally, don't have any issues with cloning the source repositories, installing pre-requisites and manually compiling four separate systems and installing them. But, it is a big turn-off for new beginners. I have a hard time recommending it to people coming from Windows background.

Having an "apt-get install icestorm-flow" kinda installment way can increase the number of people willing to try the environment. 

Yes I know that even compiling the whole toolchain from the sources is way easier and have less problems than installing ISE / Vivado or any other commercial toolchain but I'm asking whether can we do even better.

Kicad has similar convenience for some time: http://kicad-pcb.org/download/ubuntu/ Users can either tap into stable or nightly build repositories and after that Kicad can be installed/updated over apt-get commands. 

If it is hard to maintain an Ubuntu package, pre-compiled all-in-one compressed releases can be good as well.  Example for this is GNU Arm toolchain. Users can download the compressed release file from https://developer.arm.com/open-source/gnu-toolchain/gnu-rm/downloads and after uncompressing and set a path, the complete toolchain is ready. 

Thanks again for all the efforts. 
ihsan.

drawkula, 2018-05-24:
Debian has packages: `apt-cache show fpga-icestorm arachne-pnr yosys`

Updating the contents probably will not be too difficult.

rqou, 2018-05-25:
I know many people disagree with me, but I vote for "pre-compiled all-in-one compressed releases." The builds I've been making (once I get them working again) are linked with musl and work across basically all distros including Android (since icestorm does not require NSS).

rqou, 2018-05-25:
Do note that some of the programs (e.g. icetime) hardcode the prefix configured at build time and will not be able to find data files if the package is "just unpacked" and not installed into the correct directory. Perhaps this can be improved.

RGD2, 2018-05-25:
If following the official instructions is too difficult for new users,
they're probably better off sticking with icestudio, which is binary and
includes a gui:

http://icestudio.readthedocs.io/en/latest/source/installation.html#windows

icestudio-0.3.3-beta-win64.exe was 105 MB. It seems to be about 263 MB on
disk after 'toolchain' installation. It includes Python2.7 and uses apio
internally, but that doesn't matter to the end-user, since the icestudio
installer is the only one that needs to be run.

It's a very 'Arduino'-like experience, which I think is what you want if
you want pre-compiled stuff.

Impressively quick too -- adding another output to one of the 'blink'
examples, and it was rebuilt and running within five seconds after hitting
'Upload'. (YMMV, my PC is plenty fast). It even includes libusbk driver
installation, which vastly improves upload speed (included in that 5
seconds!)

apio is good, and works wherever pip does.
If you want prebuilt binaries, it's the way to go, IM(NS)HO.

Otherwise there's not much getting around living on the bleeding edge git
versions.

Here's something that may help, which may be especially good if the
'official' steps look too hard and one doesn't want to manually setup a VM
and manually download and 'install' a Linux Distro on a windows PC.

https://github.com/RGD2/icestorm-git-sing

This provides a (relatively) easy way to build the toolchain from scratch
under windows, in such a way as to be useful for submitting tickets,
confirming bugs etc.

Requires download and installation of three or four (free software) windows
installers that have nothing to do with icestorm, but little to no config
effort afterwards.

Takes perhaps 15? GB of disk space, and 13 minutes to do a full rebuild for
me. A 'make clean' when you're done zaps the whole compressed binary image
to save disk space.
And then you can delete the VM too, if you want. Redoing it all doesn't
take many steps.

If you keep the ~540MB image around, it's pretty quick to load.

I've confirmed that icotools (included) can program an icoboard on the
ico-usb adaptor on a windows 10 machine as well.

In fact, it's not such a bad idea to do something like this even if you're
on other systems / distros, just to eliminate your dependencies and
environment as variables for hunting bugs.

kehribar, 2018-05-25:
It seems Apio project maintains "compressed folder" toolchain only release. They are not 'nightly builds' but it should do the trick. 

https://github.com/FPGAwars/toolchain-icestorm/releases

After unpacking the necessary release file into a directory and adding prefixes to Makefile for toolchain programs for that specific folder, example projects compiled perfectly without messing with $PATH.

Thanks everyone for suggestions.

I'm closing the issue.

closed 2018-05-25 by kehribar