YosysHQ/icestorm #141: iceprog does not find iCEstick on Windows

soccermitchy, 2018-05-25:
Hello, I got a new iCEstick today, trying to use it with the icestorm toolchain on Windows. I can get a bitstream generated fine, but when I do `iceprog my_bitstream.bin` I just get this:
```init..
Can't find iCE FTDI USB device (vendor_id 0x0403, device_id 0x6010).
ABORT.
```

I'm thinking this is a bug in icestorm/iceprog itself, since Diamond Programmer off the Lattice site can program the iCEstick with no problem. I also double checked what device/vendor IDs it's showing up as, and they matched 0x0403/0x6010, as shown in this screenshot.
[image]

How would I fix this? I even tried with `-d i:0x0403:0x6010`, although that didn't fix it at all.

cr1901, 2018-05-25:
`LibFTDI` doesn't really play nice w/ Windows by default. Unlike Linux, it's not possible for `libusb`/`libFTDI` to switch between a serial port USB device and FTDI functionality at will; you need a special driver for this.

... and @daveshah1 beat me to it :).

daveshah1, 2018-05-25:
For custom boards such as the icebreaker (cc @esden), I wonder if this could be fixed by disabling VCP in the EEPROM for the first port? Or does Windows need to be explicitly told to use libusb, as opposed to just not autobinding the serial port?

closed 2018-05-25 by cliffordwolf

josyb, 2018-05-25:
This is rather an issue for the manufacturer of the iCEstick who should have put in his own Vendor ID and Product ID instead of relying on the generic FTDI IDs.

RGD2, 2018-05-25:
Eh, If windows users stick to apio-ide or icestudio then they're covered
for this, especially the latter as it includes Zadig and prompts the user
with sensible instructions.
Although note that I have had a fail-with-error if more than one FTDIchip
device is plugged in at once on windows 10, dies with  ftdi_usb_get_strings
failed: -4 (libusb_open() failed).
Might be because it seems windows 10 handles USB drivers differently, it
doesn't seem to assign drivers per port like windows 7 and earlier did.
(where moving a known device to a different port used to result in windows
going through the whole 'installing driver' BS all over again. One could
have a different driver for the same device depending on which port one
plugged it into, but that's not how every other OS (and common sense!)
would have it.

Also fixed if windows users just run iceprog from a supported distro within
a VM (and remember to let the VM connect the USB device).

josyb, 2018-05-25:
@RGD2 ... my observation still holds ...

soccermitchy, 2018-05-25:
Well, running Zadig didn't work. It shows the device using the WinUSB driver (as shown in the screenshot below), but I still get the same error. Maybe re-open this issue? Here's a screenshot of Zadig (and it shows the same thing for `Interface 1` on the cable):
[image]

daveshah1, 2018-05-25:
As shown in the linked instructions, you need to select libusbK not WinUSB.

soccermitchy, 2018-05-25:
Ahh, I missed that part and just assumed WinUSB would work (I've used Zadig for other stuff in the past so I assumed it was the same)

soccermitchy, 2018-05-25:
Ok, just tried it (and rebooted because, well, Windows) with libusbK, still nothing...
[image]

soccermitchy, 2018-05-25:
I just tried that (originally was using binaries from the platformio `toolchain-icestorm` package), getting this error now (same thing, just a different message I'm assuming?)
```F:\Misc\code\iceStick\UsbDmx>iceprog hardware.bin
init..
Can't find iCE FTDI USB device (vendor_id 0x0403, device_id 0x6010 or 0x6014).
ABORT.```

daveshah1, 2018-05-25:
Yes, just a different message from a newer version. Not sure what to suggest now, other than a Linux VM (make sure you enable USB 2.0 in that case). Otherwise maybe @cr1901 or @mmicko can help?

soccermitchy, 2018-05-25:
I would have done that originally, tbh, except I'm using the free version of the platformio tools, and the remote uploading stuff (which is that I'd need to use it with a VM to upload) only supports 50 uploads per month on the free version. Heck, I could use an actual Linux laptop I have to do it, except it's relatively underpowered to run much graphical stuff which is why I'm doing this all on Windows.

soccermitchy, 2018-05-25:
Until a solution is found, is there a way to automate using Diamond Programmer, ie. from a `iceprog.bat` file so I can get platformio to call it?

mmicko, 2018-05-25:
@soccermitchy I had a same problem on laptop under Windows. Thing is that not all USB ports are same, some goes trough additional internal USB habs so try connecting on different port. Also if you have USB2 ports (black) and was trying using it on USB3 (blue) try moving to USB2 that could help as well. 

Noticed that under Virtual Box with extensions and USB2/3 support enabled, all works fine when using from Linux and same port not usable under Windows. So it seams that libusbk driver is not working in all cases.

soccermitchy, 2018-05-25:
I'm having the same issue on both USB2 ports I have that aren't in use right now, but Diamond works with my USB3 hub perfectly fine. I can use it - regardless of the port - perfectly fine from VMWare with Linux, but that's not much of an option right now unless I go for the platformio paid versions since programming under a VM is a pain in the ass, even with vmware tools.

soccermitchy, 2018-05-25:
So... bit of a workaround I made. I renamed `iceprog.exe` to something else not ending in `.exe` and made an `iceprog.bat` with these contents: 
```@echo off
powershell %~dp0\modxcf.ps1 %~dp0\template.xcf %1 %~dp0\new.xcf
"F:\lscc\Programmer\3.10_x64\bin\nt64\pgrcmd.exe" -infile %~dp0\new.xcf
rm %~dp0\new.xcf
```
Next to that, I had a file named `modxcf.ps1` with these contents:
```
param ($sourceXCF, $bitstream, $output)

$xml = (Get-Content $sourceXCF)
$xml | % { $_.Replace("FILENAMEHERE", $bitstream) } | Set-Content $output
```

With those two XML files in place, I made a file named `template.xcf` in Diamond, and saved it in that directory. After that, I closed Diamond, opened template.xcf in Notepad, and replaced all the bitstream paths with `FILENAMEHERE`. Once you do that, you can just do `iceprog <bitstream_path>` and it'll work fine. Terrible workaround, I know, but it's (hopefully) only temporary til I can get `iceprog` to work.

soccermitchy, 2018-05-27:
Some more info that may help in getting the issue fixed: I just found `find_all.exe` from FPGAwars (grabbed it from [here](https://github.com/FPGAwars/libftdi-cross-builder/releases)) and ran it, and got this output:
```
Number of FTDI devices found: 3
Checking device: 0
ftdi_usb_get_strings failed: -4 (libusb_open() failed)
```

After looking up the return code, I found out in the [libftdi docs that I found](https://www.intra2net.com/en/developer/libftdi/documentation/group__libftdi.html#ga17e8eae09e58bd3bfeeab8decfce7fd9) that the return code `-4` means `unable to open device`... Tried running find_all as an administrator, thinking it was a permissions issue, but I didn't get any different results.

Also, since the issue still isn't fixed, can it be re-opened?

soccermitchy, 2018-05-27:
The more I look into this issue, the more unsure of I am of where exactly the issue is... Right now, I'm thinking it's in `libftdi`, since `find_all.exe` which uses `libftdi` doesn't work, but the `xusb.exe` example from `libusb` does work. Specifically, I think it's in `ftdi_usb_get_strings`'s `libusb_open` call, specifically [on line 420 of ftdi.c](https://www.intra2net.com/en/developer/libftdi/documentation/ftdi_8c_source.html#l00420), since the open call there is what's failing as far as I can tell (well, the error from find_all.exe is a pretty big giveaway). On the other hand, `xusb` is using `libusb_open_device_with_vid_pid`, which does a `libusb_open` call internally, which does work. With this info, I'm thinking that maybe an invalid device is being passed to `libusb_open` in `libftdi`? I'm honestly unsure of what to do here.

1138-4EB, 2019-08-20:
> [@RGD2]
> Also fixed if windows users just run iceprog from a supported distro within
a VM (and remember to let the VM connect the USB device).

I used a Docker Desktop container, which runs on top of a Hyper-V VM on Windows 10. Although sharing USB devices with Hyper-V VMs is not supported yet, I could successfully use USB/IP by building and loading the required kernel modules. A shell script and instructions to configure an icestick with `iceprog` from image `ghdl/synth:icestorm` are available at https://github.com/ghdl/docker/tree/master/usbip. I hope it can be useful to someone having issues with Zadig/libusbK/WinUSB who has docker-for-win or docker-for-mac installed already.

drayde, 2019-09-20:
Had the same issue with my icebreaker board, using Zadig with libusbK worked for me.

kyakpirate, 2020-03-08:
I had the same issue on Windows 10.   Zadiag would communicate with my Lattice device with USB ID 0403:6010 but when I tried to run iceprog, it did not communicate.  I had my USB cable directly  connected to a USB port on my computer.   It did not work.   I got this hub,  Atolla USB 3.0 hub, and connected my device through a port on it.   Then the Zadiag and iceprog programs worked the 1st time.

kyakpirate, 2020-03-08:
Opps, I wrote Zadiag insteadl of Zadig above.

AnasMalas, 2025-04-10:
Adding this here in case someone finds this issue in the future. There is now another pretty easy way to do this. 

Installation and setup:
- Install WSL on your windows machine, install iceprog there. I used Ubuntu
- Download and install https://github.com/dorssel/usbipd-win. More info here: https://learn.microsoft.com/en-us/windows/wsl/connect-usb
- Using usbipd, see the list of devices, find the FTDI device and force bind it (binding requires admin access, but should be a one time thing).

Use
- Every time the device is plugged in, use usbipd to attach it to WSL
- Your device now shows up in WSL and can be programmed with iceprog directly

Enjoy :)