YosysHQ/icestorm #179: Converting ASCI-TXT file to .blif

qaarah, 2018-07-31:
Hi, 
I want to convert m .asc file to .blif file before writing it to the a binary. 
My .asc is an output from arachne-pnr. That is then used by icepack to convert to binary. Is there any way that before getting binary, I can convert the .asc or .txt file to a blif file?
Thanks

daveshah1, 2018-07-31:
There is no direct .asc to .blif flow. You could use `icetime` to generate a Verilog timing netlist from the .asc and use Yosys to convert that to BLIF.

You could also use `icebox_vlog` to generate functional Verilog from the .asc and convert that to BLIF with Yosys, but in this case it would not correspond to the original design netlist, but merely be functionally equivalent to it.

qaarah, 2018-07-31:
Thanks Dave, 
I used the same way you mentioned, i.e using icebox_vlog to generate functional verilog. ( I can not see any command to dump this verilog, is there any?) I used the obtained verilog to generate blif using yosys. 
But actually, I want this to give this blif to VPR again, to generate an output, (.net, .p or .r) that then my tool flow can understand. any help will be appreciated, 
Thanks

daveshah1, 2018-07-31:
I am not sure what you mean about "dump this verilog" with regards to icebox_vlog.

In theory, I think the Verilog timing netlists produced by icetime are the closest to what you want.

qaarah, 2018-07-31:
Regarding icebox_vlog, dumping means to save the output Verilog directly and accesing for the new flow. (This would not be a problem). Thanks, 
Yes, that could be the possibility and I will use this and will let you know. 
Thanks again,

daveshah1, 2018-07-31:
Like any command line tool, you can redirect the output to a file using ">"
e.g.
```
icebox_vlog design.asc > design_out.v
```

qaarah, 2018-07-31:
Yes, thats fine. "icetime" generates the timing analysis report. Right? do you think I can use it for VPR as an input? I mean, as (.net file)?

daveshah1, 2018-07-31:
You can use `icetime -o out.v` to generate a Verilog timing netlist. I don't know whether or not that is useful for your unknown flow.

qaarah, 2018-11-05:
Hi @daveshah1 
Considering the thread above I have a relevant question here. Its about converting the `ASCI` file to `verilog` using `icebox_vlog`.  The names of inputs (wires) and outputs has been changed in the ouput verilog using  `icebox_vlog`.  For example in my case I use the following example.
`module top (
	input  clk,
	input  RX,
	output TX,
	output reg LED1,
	output reg LED2,
	output reg LED3,
	output reg LED4,
	output reg LED5
);
	parameter integer BAUD_RATE = 9600;
	parameter integer CLOCK_FREQ_HZ = 12000000;
	localparam integer HALF_PERIOD = CLOCK_FREQ_HZ / (2 * BAUD_RATE);

	reg [7:0] buffer;
	reg buffer_valid;

	reg [$clog2(3*HALF_PERIOD):0] cycle_cnt;
	reg [3:0] bit_cnt = 0;
	reg recv = 0;

	initial begin
		LED1 = 0;
		LED2 = 0;
		LED3 = 0;
		LED4 = 0;
		LED5 = 0;
	end

	always @(posedge clk) begin
		buffer_valid <= 0;
		if (!recv) begin
			if (!RX) begin
				cycle_cnt <= HALF_PERIOD;
				bit_cnt <= 0;
				recv <= 1;
			end
		end else begin
			if (cycle_cnt == 2*HALF_PERIOD) begin
				cycle_cnt <= 0;
				bit_cnt <= bit_cnt + 1;
				if (bit_cnt == 9) begin
					buffer_valid <= 1;
					recv <= 0;
				end else begin
					buffer <= {RX, buffer[7:1]};
				end
			end else begin
				cycle_cnt <= cycle_cnt + 1;
			end
		end
	end

	always @(posedge clk) begin
		if (buffer_valid) begin
			if (buffer == "1") LED1 <= !LED1;
			if (buffer == "2") LED2 <= !LED2;
			if (buffer == "3") LED3 <= !LED3;
			if (buffer == "4") LED4 <= !LED4;
			if (buffer == "5") LED5 <= !LED5;
		end
	end

	assign TX = RX;
endmodule ` 
With following commands 
`yosys -p "synth_ice40 -blif rs.blif" rs232demo.v`
`arachne-pnr -o rs.asc rs.blif`
`icebox_vlog rs.asc > rs_output.v`
Where the output verilog is as follows.
`module chip (input io_7_17_0, output io_0_2_0, input io_0_9_0, output io_0_14_1, output io_5_17_1, output io_5_17_0, output io_6_17_1, output io_13_14_0);

wire n1;
// (0, 0, 'glb_netwk_0')
// (5, 0, 'logic_op_tnr_4')
// (5, 1, 'neigh_op_rgt_4')
// (5, 2, 'neigh_op_bnr_4')
// (5, 4, 'lutff_global/s_r')
// (5, 6, 'lutff_global/s_r')
// (5, 7, 'lutff_global/s_r')
// (6, 0, 'logic_op_top_4')
// (6, 1, 'lutff_4/out')
// (6, 2, 'neigh_op_bot_4')
// (6, 4, 'lutff_global/s_r')
// (6, 5, 'lutff_global/s_r')
// (7, 0, 'fabout')
// (7, 0, 'local_g1_4')
// (7, 0, 'logic_op_tnl_4')
// (7, 1, 'neigh_op_lft_4')
// (7, 2, 'neigh_op_bnl_4')
// (7, 5, 'lutff_global/s_r')
// (7, 7, 'lutff_global/s_r')
// (8, 4, 'lutff_global/s_r')

wire io_7_17_0;
// (0, 0, 'glb_netwk_1')
// (1, 14, 'lutff_global/clk')
// (5, 4, 'lutff_global/clk')
// (5, 6, 'lutff_global/clk')
// (5, 7, 'lutff_global/clk')
// (5, 14, 'lutff_global/clk')
// (5, 15, 'lutff_global/clk')
// (5, 17, 'span4_horz_r_0')
// (6, 4, 'lutff_global/clk')`
Note: I pasted shorter version of output here due to space. 
Could you please tell the reason why does the names of inputs and outputs change? Is there any way to keep them the same as was in original file. 
The change in name could lead to the problem in equivalence checking. 
Thanks