YosysHQ/icestorm #186: arachne-pnr couldn't process wide bit width counter for lm4k

BravoFpga, 2018-09-17:
When I compile following rtl, 

    `timescale 1ns/1ps
    module cnt (CLK,A1);
    input  CLK;
    output reg A1; 

    //parameter LENGTH = 'd153; // OKay
    parameter LENGTH = 'd154; // Bad
    reg [LENGTH-1:0]    cnt;

    always @(posedge CLK) cnt <= cnt + 1'b1;
    always @(posedge CLK) A1 <= cnt[LENGTH-1];

    endmodule
the arachne-pnr output assert message and abort program as follows.

    rachne-pnr -d lm4k -P swg25tr -p cnt.pcf cnt.blif -o cnt.txt
    seed: 1
    device: lm4k
    read_chipdb +/share/arachne-pnr/chipdb-lm4k.bin...
      supported packages: cm36, cm49, swg25tr
    read_blif cnt.blif...
    prune...
    read_pcf cnt.pcf...
    instantiate_io...
    pack...
    
    After packing:
    IOs          2 / 18
    GBs          0 / 8 
      GB_IOs     0 / 8 
    LCs          157 / 3520
      DFF        4
      CARRY      2   
      CARRY, DFF 151 
      DFF PASS   1   
      CARRY PASS 1
    BRAMs        0 / 20
    WARMBOOTs    0 / 1 
    PLLs         0 / 0 
    
    place_constraints...
    promote_globals...
      promoted CLK$2, 155 / 155 
      promoted 1 nets
        1 clk 
      1 globals
        1 clk 
    realize_constants...
    place...
      initial wire length = 176 
    arachne-pnr: src/vector.hh:84: BasedVector<T, B>::reference BasedVector<T, B>::operator[](BasedVector<T, B>::size_type) [with T = int; typename std::vector<_RealType>::size_type B = 1; BasedVector<T, B>::reference = int&; BasedVector<T, B>::size_type = long unsigned int]: Assertion `i >= B && i < B + v.size()' failed.

cliffordwolf, 2018-09-17:
Hi! The arachne-pnr issue tracker is at this link: https://github.com/YosysHQ/arachne-pnr/issues/new

closed 2018-09-17 by cliffordwolf

BravoFpga, 2018-09-17:
ouch!
thank you for indicate it.
l`ll re register it to right place in later.

-------- オリジナルのメッセージ --------
送信元:Clifford Wolf <notifications@github.com>
送信済み:Mon, 17 Sep 2018 20:32:37 +0900
送信先:cliffordwolf/icestorm <icestorm@noreply.github.com>
Cc:BravoFpga <hc16804@yahoo.co.jp>,Author <author@noreply.github.com>
件名:Re: [cliffordwolf/icestorm] arachne-pnr couldn't process wide bit width counter for lm4k (#186)