YosysHQ/icestorm #189: Problem building nextpnr on Ubuntu 18.04

mplichota, 2018-11-14:
*** nextpnr build attempt
myron@myron-HP-Pavilion-dm4-Notebook-PC:~/Mystorm$ git clone https://github.com/YosysHQ/nextpnr nextpnr
Cloning into 'nextpnr'...
remote: Enumerating objects: 224, done.
remote: Counting objects: 100% (224/224), done.
remote: Compressing objects: 100% (111/111), done.
remote: Total 10379 (delta 155), reused 162 (delta 113), pack-reused 10155
Receiving objects: 100% (10379/10379), 4.49 MiB | 1.95 MiB/s, done.
Resolving deltas: 100% (7579/7579), done.
myron@myron-HP-Pavilion-dm4-Notebook-PC:~/Mystorm$ cd nextpnr
myron@myron-HP-Pavilion-dm4-Notebook-PC:~/Mystorm/nextpnr$ cmake -DARCH=ice40 -DCMAKE_INSTALL_PREFIX=/usr/local .
-- The C compiler identification is GNU 7.3.0
-- The CXX compiler identification is GNU 7.3.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Found PythonInterp: /usr/bin/python3 (found suitable version "3.6.6", minimum required is "3.5") 
-- Found PythonLibs: /usr/lib/x86_64-linux-gnu/libpython3.6m.so (found suitable version "3.6.6", minimum required is "3.5") 
-- Looking for pthread.h
-- Looking for pthread.h - found
-- Looking for pthread_create
-- Looking for pthread_create - not found
-- Looking for pthread_create in pthreads
-- Looking for pthread_create in pthreads - not found
-- Looking for pthread_create in pthread
-- Looking for pthread_create in pthread - found
-- Found Threads: TRUE  
CMake Error at /usr/share/cmake-3.10/Modules/FindBoost.cmake:1947 (message):
  Unable to find the requested Boost libraries.

  Boost version: 1.65.1

  Boost include path: /usr/include

  Could not find the following Boost libraries:

          boost_filesystem
          boost_thread
          boost_program_options

  Some (but not all) of the required Boost libraries were found.  You may
  need to install these additional Boost libraries.  Alternatively, set
  BOOST_LIBRARYDIR to the directory containing Boost libraries or BOOST_ROOT
  to the location of Boost.
Call Stack (most recent call first):
  CMakeLists.txt:83 (find_package)

-- Found OpenGL: /usr/lib/x86_64-linux-gnu/libOpenGL.so   
CMake Warning at /usr/share/cmake-3.10/Modules/FindBoost.cmake:1626 (message):
  No header defined for python-py366; skipping header check
Call Stack (most recent call first):
  CMakeLists.txt:124 (find_package)

-- Could NOT find Boost
CMake Warning at /usr/share/cmake-3.10/Modules/FindBoost.cmake:1626 (message):
  No header defined for python-py36; skipping header check
Call Stack (most recent call first):
  CMakeLists.txt:131 (find_package)

-- Could NOT find Boost
CMake Warning at /usr/share/cmake-3.10/Modules/FindBoost.cmake:1626 (message):
  No header defined for python-py3; skipping header check
Call Stack (most recent call first):
  CMakeLists.txt:131 (find_package)

-- Could NOT find Boost
CMake Warning at /usr/share/cmake-3.10/Modules/FindBoost.cmake:1626 (message):
  No header defined for python3; skipping header check
Call Stack (most recent call first):
  CMakeLists.txt:141 (find_package)

-- Could NOT find Boost
CMake Warning at /usr/share/cmake-3.10/Modules/FindBoost.cmake:1626 (message):
  No header defined for python36; skipping header check
Call Stack (most recent call first):
  CMakeLists.txt:148 (find_package)

-- Could NOT find Boost
CMake Warning at /usr/share/cmake-3.10/Modules/FindBoost.cmake:1626 (message):
  No header defined for python37; skipping header check
Call Stack (most recent call first):
  CMakeLists.txt:155 (find_package)

-- Could NOT find Boost
CMake Warning at /usr/share/cmake-3.10/Modules/FindBoost.cmake:1626 (message):
  No header defined for python-3.6; skipping header check
Call Stack (most recent call first):
  CMakeLists.txt:163 (find_package)

-- Could NOT find Boost
CMake Error at CMakeLists.txt:170 (message):
  No version of Boost::Python 3.x could be found.

-- Configuring incomplete, errors occurred!
See also "/home/myron/Mystorm/nextpnr/CMakeFiles/CMakeOutput.log".
See also "/home/myron/Mystorm/nextpnr/CMakeFiles/CMakeError.log".
myron@myron-HP-Pavilion-dm4-Notebook-PC:~/Mystorm/nextpnr$ make -j$(nproc)
make: *** No targets specified and no makefile found.  Stop.
myron@myron-HP-Pavilion-dm4-Notebook-PC:~/Mystorm/nextpnr$ 
*** end nextpnr build attempt

*** search for all things boost
myron@myron-HP-Pavilion-dm4-Notebook-PC:~/Mystorm/nextpnr$ find /usr -iname '*boost*'
/usr/include/boost
/usr/include/boost/serialization/boost_unordered_map.hpp
/usr/include/boost/serialization/boost_unordered_set.hpp
/usr/include/boost/serialization/boost_array.hpp
/usr/include/boost/geometry/geometries/adapted/boost_polygon
/usr/include/boost/geometry/geometries/adapted/boost_array.hpp
/usr/include/boost/geometry/geometries/adapted/boost_polygon.hpp
/usr/include/boost/geometry/geometries/adapted/boost_fusion.hpp
/usr/include/boost/geometry/geometries/adapted/boost_range
/usr/include/boost/geometry/geometries/adapted/boost_tuple.hpp
/usr/include/boost/icl/detail/boost_config.hpp
/usr/include/boost/fusion/include/boost_array.hpp
/usr/include/boost/fusion/include/boost_tuple.hpp
/usr/include/boost/fusion/adapted/boost_array
/usr/include/boost/fusion/adapted/boost_array.hpp
/usr/include/boost/fusion/adapted/boost_tuple
/usr/include/boost/fusion/adapted/boost_tuple/boost_tuple_iterator.hpp
/usr/include/boost/fusion/adapted/boost_tuple.hpp
/usr/include/boost/convert/detail/boost_parameter_ext.hpp
/usr/include/boost/hana/ext/boost
/usr/include/boost/hana/ext/boost.hpp
/usr/include/boost/chrono/typeof/boost
/usr/include/boost/type_index/runtime_cast/boost_shared_ptr_cast.hpp
/usr/include/c++/7/bits/boost_concept_check.h
/usr/lib/x86_64-linux-gnu/libboost_date_time.so.1.65.1
/usr/lib/x86_64-linux-gnu/libboost_system.so.1.65.1
/usr/lib/x86_64-linux-gnu/libboost_iostreams.so.1.65.1
/usr/lib/x86_64-linux-gnu/libboost_filesystem.so.1.65.1
/usr/lib/x86_64-linux-gnu/libboost_thread.so.1.65.1
/usr/lib/x86_64-linux-gnu/libboost_locale.so.1.65.1
/usr/share/lintian/overrides/libboost-thread1.65.1
/usr/share/lintian/overrides/libboost1.65-dev
/usr/share/lintian/overrides/libboost-date-time1.65.1
/usr/share/lintian/overrides/libboost-filesystem1.65.1
/usr/share/lintian/overrides/libboost-system1.65.1
/usr/share/lintian/overrides/libboost-iostreams1.65.1
/usr/share/lintian/overrides/libboost-locale1.65.1
/usr/share/doc/libboost-thread1.65.1
/usr/share/doc/libboost1.65-dev
/usr/share/doc/libboost-date-time1.65.1
/usr/share/doc/libboost-filesystem1.65.1
/usr/share/doc/libboost-dev
/usr/share/doc/libboost-system1.65.1
/usr/share/doc/libboost-iostreams1.65.1
/usr/share/doc/libboost-locale1.65.1
/usr/share/cmake-3.10/Modules/FindBoost.cmake
/usr/share/cmake-3.10/Help/module/FindBoost.rst
*** end search for all things boost

swetland, 2018-11-17:
I believe this resolved the boost dependencies for me on 18.04:

`$ sudo apt-get install libboost-all-dev`

mplichota, 2018-11-18:
Looks like that did the trick. Thank you Sir!

closed 2018-11-18 by mplichota