YosysHQ/icestorm #192: Raspberry Pi support? (error compiling in icepack.cc)

gojimmypi, 2018-12-14:
I tried to install icestorm on a fresh and updated Raspberry Pi (Linux raspberrypi 4.14.79-v7+ #1159 SMP Sun Nov 4 17:50:20 GMT 2018 armv7l)  using the instructions [here](http://www.clifford.at/icestorm/#install) first using the Ubuntu prerequisites like this:
```
sudo apt-get install build-essential clang bison flex libreadline-dev \
                     gawk tcl-dev libffi-dev git mercurial graphviz   \
                     xdot pkg-config python python3 libftdi-dev \
                     qt5-default python3-dev libboost-dev
``` 
When I try to `make -j$(nproc)` I receive a bunch of compiler errors:

```
0 $ cd workspace
pi@raspberrypi : ~/workspace
0 $ git clone https://github.com/cliffordwolf/icestorm.git icestorm
Cloning into 'icestorm'...
remote: Enumerating objects: 3473, done.
remote: Total 3473 (delta 0), reused 0 (delta 0), pack-reused 3473
Receiving objects: 100% (3473/3473), 2.17 MiB | 2.22 MiB/s, done.
Resolving deltas: 100% (2308/2308), done.
pi@raspberrypi : ~/workspace
0 $ cd icestorm
pi@raspberrypi : ~/workspace/icestorm
0 $ make -j$(nproc)
make -C icebox all
make -C icepack all
make -C iceprog all
make -C icemulti all
make[1]: Entering directory '/home/pi/workspace/icestorm/iceprog'
make[1]: Entering directory '/home/pi/workspace/icestorm/icepack'
make[1]: Entering directory '/home/pi/workspace/icestorm/icebox'
g++ -MD -O2  -Wall -std=c++11 -I/usr/local/include   -c -o icepack.o icepack.cc
python3 icebox_chipdb.py -3 > chipdb-384.new
make[1]: Entering directory '/home/pi/workspace/icestorm/icemulti'
g++ -MD -O2  -Wall -std=c++11 -I/usr/local/include   -c -o icemulti.o icemulti.cc
cc -MD -O2  -Wall -std=c99 -I/usr/local/include    -c -o iceprog.o iceprog.c
cc -o iceprog  iceprog.o -L/usr/local/lib -lm -lftdi
make[1]: Leaving directory '/home/pi/workspace/icestorm/iceprog'
python3 icebox_chipdb.py > chipdb-1k.new
icepack.cc: In member function ‘void FpgaConfig::write_ascii(std::ostream&) const’:
icepack.cc:53:62: warning: format ‘%lu’ expects argument of type ‘long unsigned int’, but argument 6 has type ‘std::vector<std::vector<bool> >::size_type {aka unsigned int}’ [-Wformat=]
 #define error(...) do { fprintf(stderr, "Error: " __VA_ARGS__); exit(1); } while (0)
                                                              ^
icepack.cc:902:6: note: in expansion of macro ‘error’
      error("cram_x %d (bit %d, %d) larger than bank size %lu\n", cram_x, bit_x, bit_y, this->cram[cram_bank].size());
      ^~~~~
icepack.cc:53:62: warning: format ‘%lu’ expects argument of type ‘long unsigned int’, but argument 7 has type ‘std::vector<bool>::size_type {aka unsigned int}’ [-Wformat=]
 #define error(...) do { fprintf(stderr, "Error: " __VA_ARGS__); exit(1); } while (0)
                                                              ^
icepack.cc:905:6: note: in expansion of macro ‘error’
      error("cram_y %d (bit %d, %d) larger than bank %d size %lu\n", cram_y, bit_x, bit_y, cram_bank, this->cram[cram_bank][cram_x].size());
      ^~~~~
icepack.cc:53:62: warning: format ‘%lu’ expects argument of type ‘long unsigned int’, but argument 6 has type ‘std::vector<std::vector<bool> >::size_type {aka unsigned int}’ [-Wformat=]
 #define error(...) do { fprintf(stderr, "Error: " __VA_ARGS__); exit(1); } while (0)
                                                              ^
icepack.cc:924:8: note: in expansion of macro ‘error’
        error("%d %d bram_x %d higher than loaded bram size %lu\n",bit_x+i, bit_y, bram_x,  this->bram[bram_bank].size());
        ^~~~~
icepack.cc:53:62: warning: format ‘%lu’ expects argument of type ‘long unsigned int’, but argument 4 has type ‘std::vector<bool>::size_type {aka unsigned int}’ [-Wformat=]
 #define error(...) do { fprintf(stderr, "Error: " __VA_ARGS__); exit(1); } while (0)
                                                              ^
icepack.cc:928:8: note: in expansion of macro ‘error’
        error("bram_y %d higher than loaded bram size %lu\n", bram_y,  this->bram[bram_bank][bram_x].size());
        ^~~~~
g++ -o icemulti  icemulti.o -lm -lstdc++
make[1]: Leaving directory '/home/pi/workspace/icestorm/icemulti'
python3 icebox_chipdb.py -8 > chipdb-8k.new
g++ -o icepack  icepack.o -lm -lstdc++
ln -sf icepack iceunpack
make[1]: Leaving directory '/home/pi/workspace/icestorm/icepack'
python3 icebox_chipdb.py -5 > chipdb-5k.new
mv chipdb-384.new chipdb-384.txt
make -C icepll all
make[1]: Entering directory '/home/pi/workspace/icestorm/icepll'
g++ -MD -O2  -Wall -std=c++11 -I/usr/local/include   -c -o icepll.o icepll.cc
g++ -o icepll  icepll.o -lm -lstdc++
make[1]: Leaving directory '/home/pi/workspace/icestorm/icepll'
python3 icebox_chipdb.py -4 > chipdb-lm4k.new

mv chipdb-1k.new chipdb-1k.txt
make -C icetime all
make[1]: Entering directory '/home/pi/workspace/icestorm/icetime'
python3 timings.py > timings.inc.new
mv timings.inc.new timings.inc
g++ -MD -O2  -Wall -std=c++11 -I/usr/local/include -DPREFIX='"/usr/local"' -DCHIPDB_SUBDIR='"icebox"'   -c -o iceutil.o iceutil.cc
g++ -MD -O2  -Wall -std=c++11 -I/usr/local/include -DPREFIX='"/usr/local"' -DCHIPDB_SUBDIR='"icebox"'   -c -o icetime.o icetime.cc
...[snip, etc]...
```
Is it reasonable to try to get icestorm working on a Raspberry Pi? Perhaps some Pi-specific prerequisites are missing?

drawkula, 2018-12-14:
Where are the errors?  
I think you only showed warnings...

drawkula, 2018-12-14:
Just test build some of the examples that come with icestorm's sources.

Did you already build arachne-pnr and yosys? Or nextpnr?

For me it worked when I tried to get started with FPGAs (last winter) but unluckily real life distracted me heavily since then. I hope to get a restart with HX8K and UP5K soon. RISCV stuff definitely needs to be explored on FPGAs. ;-)

mmicko, 2021-09-06:
Those warnings were safe to ignore on ARM.

closed 2021-09-06 by mmicko