YosysHQ/icestorm #194: iceprog leaves FTDI device in odd mode

osresearch, 2018-12-15:
On the upduino v2 with ice40 up5k and an FTDI 232H chip, the device enumerates as a serial device and creates `/dev/ttyUSB0`.  I can send/receive data from the board fine:

```
[1455866.817013] usb 1-1: new high-speed USB device number 90 using xhci_hcd
[1455866.957203] usb 1-1: New USB device found, idVendor=0403, idProduct=6014
[1455866.957211] usb 1-1: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[1455866.957216] usb 1-1: Product: Single RS232-HS
[1455866.957220] usb 1-1: Manufacturer: FTDI
[1455866.958328] ftdi_sio 1-1:1.0: FTDI USB Serial Device converter detected
[1455866.958487] usb 1-1: Detected FT232H
[1455866.958846] usb 1-1: FTDI USB Serial Device converter now attached to ttyUSB0
```

When the board is reprogrammed with `iceprog`, however, it disconnects from `/dev/ttyUSB0`:

```
[1455919.013105] ftdi_sio ttyUSB0: FTDI USB Serial Device converter now disconnected from ttyUSB0
[1455919.013132] ftdi_sio 1-1:1.0: device disconnected
```

In order to have it re-enumerate and see the serial output from the FPGA, I have to unplug/re-plug the cable.  Is there a way to have `iceprog` switch the FTDI back into serial mode after programming?

osresearch, 2018-12-15:
As a workaround, I'm able to add this to my `Makefile` for forcing a reset, flashing the device, and then re-attaching the `/dev/ttyUSB0` port to the FTDI SIO kernel driver:
```
%.flash: %.bin
        iceprog -e 128 # Force a reset to stop any serial output
        iceprog $<
        echo 1-1:1.0 > /sys/bus/usb/drivers/ftdi_sio/bind  # re-attach serial
```

AmeyaVS, 2025-05-24:
I tried used  the mentioned echo command but it doesn't seem to work.
I am getting permission denied error when trying to open the bind file even with sudo.
Does anyone have a suggestion on how to fix this issue?,
For now, I need to keep unplugging and plugging in the board, for it to work.