YosysHQ/icestorm #204: icetime: No path found

gojimmypi, 2019-03-10:
when generating an `icetime` rpt file like this:

```
icetime -d hx8k -P tq144 -p syn/myfile.pcf -mtr myfile.rpt myfile.asc
```

`icetime` returns an error: `No path found!`; The `myfile.asc` from `nextpnr` exists, and same error occurs regardless if `myfile.rpt` already exists or not.

```
// Reading input .pcf file..
// Reading input .asc file..
// Reading 8k chipdb file..
// Creating timing netlist..
// Timing estimate: 0.00 ns (inf MHz)
No path found!
```

The source code shows that message in [double report(std::string n = std::string())](https://github.com/cliffordwolf/icestorm/blob/master/icetime/icetime.cc#L928), acting as if no rpt file string was specified, but there must have been as a the `myfile.rpt` is generated despite the errors.

In this a known problem? There does not appear to be an open issue - but the only other google search result for `icetime "No path found!"` was found on a bounty page [here](https://www.bountysource.com/teams/symbiflow/issues?tracker_ids=81429368).

daveshah1, 2019-03-10:
Is it possible that this is an empty design? That would be the normal cause of this message, and corresponding "infinite" maximum frequency.

Otherwise, can you post sufficient information to reproduce this message (asc, verilog and pcf).

gojimmypi, 2019-03-10:
This all started with me learning FPGA programming and looking at an [AD/DA](https://www.ebay.com/itm/221755565048) converter, and specifically [this project](https://github.com/jhol/otl-modem) mentioned [here](https://opentechlab.org.uk/videos:014:notes) as a passthru example to get started. (I have more details for the AD/DA board on [my blog](https://gojimmypi.blogspot.com/2018/06/ad9280-ad9708-ad-da-module-for-ax.html)).  

My `nextpnr` output is [here](https://gist.github.com/gojimmypi/a431af22f1764dfdab8c882df4df1f7f). There are a lot of zeros, so perhaps indeed it is empty? I renamed everything to `myfile` to ensure the dashes in the filenames were not getting confused with commandline parameters.

If indeed this is due to an empty `asc` file - that would be a great feature to add to `icetime` say that. :)

The specific constraint file they are using is [here](https://github.com/jhol/otl-modem/blob/master/fw/syn/otl-modem.pcf) with [top.v](https://github.com/jhol/otl-modem/blob/master/fw/syn/otl-modem.pcf), but copied here for this issue:

verilog:
```
/*
 * Copyright (c) 2018 Joel Holdsworth <joel@airwebreathe.org.uk>
 * All rights reserved.
 *
 * Redistribution and use in source and binary forms, with or without
 * modification, are permitted provided that the following conditions are met:
 *
 * 1. Redistributions of source code must retain the above copyright notice,
 *    this list of conditions and the following disclaimer.
 * 2. Redistributions in binary form must reproduce the above copyright
 *    notice, this list of conditions and the following disclaimer in the
 *    documentation and/or other materials provided with the distribution.
 * 3. Neither the name of copyright holder nor the names of its
 *    contributors may be used to endorse or promote products derived from
 *    this software without specific prior written permission.
 *
 * THIS SOFTWARE IS PROVIDED BY THE COPYRIGHT HOLDERS AND CONTRIBUTORS "AS IS"
 * AND ANY EXPRESS OR IMPLIED WARRANTIES, INCLUDING, BUT NOT LIMITED TO, THE
 * IMPLIED WARRANTIES OF MERCHANTABILITY AND FITNESS FOR A PARTICULAR PURPOSE
 * ARE DISCLAIMED. IN NO EVENT SHALL THE COPYRIGHT OWNER OR CONTRIBUTORS BE
 * LIABLE FOR ANY DIRECT, INDIRECT, INCIDENTAL, SPECIAL, EXEMPLARY, OR
 * CONSEQUENTIAL DAMAGES (INCLUDING, BUT NOT LIMITED TO, PROCUREMENT OF
 * SUBSTITUTE GOODS OR SERVICES; LOSS OF USE, DATA, OR PROFITS; OR BUSINESS
 * INTERRUPTION) HOWEVER CAUSED AND ON ANY THEORY OF LIABILITY, WHETHER IN
 * CONTRACT, STRICT LIABILITY, OR TORT (INCLUDING NEGLIGENCE OR OTHERWISE)
 * ARISING IN ANY WAY OUT OF THE USE OF THIS SOFTWARE, EVEN IF ADVISED OF THE
 * POSSIBILITY OF SUCH DAMAGE.
 */

module top(clk_100mhz, adclk, addb, daclk, dadb);
parameter ClkFreq = 50000000; // Hz

input clk_100mhz;
output adclk;
input [7:0] addb;
output daclk;
output [7:0] dadb;

// Clock Generator
wire clk_50mhz;
wire pll_locked;

SB_PLL40_PAD #(
  .FEEDBACK_PATH("SIMPLE"),
  .DELAY_ADJUSTMENT_MODE_FEEDBACK("FIXED"),
  .DELAY_ADJUSTMENT_MODE_RELATIVE("FIXED"),
  .PLLOUT_SELECT("GENCLK"),
  .FDA_FEEDBACK(4'b1111),
  .FDA_RELATIVE(4'b1111),
  .DIVR(4'b0000),
  .DIVF(7'b0000111),
  .DIVQ(3'b100),
  .FILTER_RANGE(3'b101)
) pll (
  .PACKAGEPIN(clk_100mhz),
  .PLLOUTGLOBAL(clk_50mhz),
  .LOCK(pll_locked),
  .BYPASS(1'b0),
  .RESETB(1'b1)
);
[... 107 lines cut ...]
#otl-modem.bin: otl-modem.asc
#	echo "Running icepak!"
#	icepack $< $@

simulate-ft2232h_interface: ft2232h_interface.vcd
	gtkwave $< >/dev/null 2>/dev/null &

ft2232h_interface-sim: \
	sim/ft2232h_interface.v \
	src/ft2232h_interface.v
	iverilog -o $@ $^

%.vcd: %-sim
	cd $(dir $<); ./$(notdir $<)

clean:
	rm -f $(PROJ).blif $(PROJ).asc $(PROJ).rpt $(PROJ).bin

.SECONDARY:
.PHONY: all prog clean

.PHONY: \
  synthesize \
  simulate-ft2232h_interface \
```

daveshah1, 2019-03-10:
This is indeed an empty design for the purposes of icetime (all it has is some disconnected input and output ports after dangling cells are swept away).

daveshah1, 2019-03-10:
FYI, the "no path found" message was added to clear up this case in #45.

gojimmypi, 2019-03-10:
Thank you for your help. I'm going to leave this issue open, with the suggestion that the `No path found` error (which sounds like a file system warning) be changed to the next noted in https://github.com/cliffordwolf/icestorm/issues/45: 

`This design is empty. It contains no paths`.

I created https://github.com/cliffordwolf/icestorm/pull/205 for this.

cliffordwolf, 2019-03-11:
Merged #205.

closed 2019-03-11 by cliffordwolf