YosysHQ/icestorm #209: Write error (single byte, rc=-110, expected 1)

jasonlee1001, 2019-03-28:
Hi,
I use Ubuntu 16.04 on VirtualBox (host system is Windows 7)
After installing all tools and the command:
```
     cd /icestorms/examples
     make
     iceprog example.bin
```

The program replys:

> init..
> cdone: high
> reset..
> cdone: low
> flash ID: 0x20 0xBA 0x16 0x10 .....
> file size: 32220
> erase 64kB sector at 0x000000..
> programming..
> Write error (single byte, rc=-110, expected 1).
> ABORT.

Can anyone know how to solve this ?

daveshah1, 2019-03-28:
I have found iceprog in Virtualbox is significantly more reliable with the extension pack installed and USB 2.0 used instead of USB 1.1 in the settings.

jasonlee1001, 2019-03-29:
> I have found iceprog in Virtualbox is significantly more reliable with the extension pack installed and USB 2.0 used instead of USB 1.1 in the settings.

Hi,
  Can you explain more details about "extension pack" ? Is it about VirtualBox ?
My laptop with USB2.0. Have you ever experience the same fault ?

RGD2, 2019-03-30:
It’s the paid add on to virtualbox, but you can trial it.

Otherwise you could get a thumb drive of a 16GB or larger size, put Ubuntu
on it, boot off it and try icestorm from within the livebooted environment.
It’ll be slow (unless it’s a really fast “USB ssd”) but it very probably
will work, because native.

Or, you could try my instructions at
https://github.com/RGD2/icestorm-git-sing

Which is how I automate building / using icestorm on windows with
virtualbox, vagrant and Singularity. (Makes it very easy to delete/replace
the VM, but you might hit the same issue on your hardware).

I’ve done a fair bit of low level USB device development on windows, and I
can report that it stinks vs any other platform. Using a VM doesn’t mean
you aren’t using MSFT’s fairly crummy/lazily written USB implementation,
and  there are many additional issues with it that just don’t exist
elsewhere.

Issues like: move the device to a different port, and it will start all
over again with driver discovery/installation, like its never seen the
device before. You can quite easily have multiple different version drivers
installed *on different ports* because you moved it to a second port, and
“updated” the driver at some point. No other OS does this, and it just
creates drama.

On Linux, by comparison, you can just tell it what driver or even actions
to take when it sees a particular vid/pid at any port.
Or if you really want you could reimplement per-port decisions like windows
too - they’re not the default though.

Most annoying is how long the “oh look at me, working so hard finding a
Driver for you” phase windows takes when new (especially USB default
profile) devices are first attached. Every competent OS just works - the
whole point of the USB standard is to predefine drivers so thing like mice,
or USB-serial or USB audio, etc etc can “just work”. But MS sat on that
standard board, and made sure that their preference of taking zero
responsibility or effort to write drivers for their platform was a-ok too.

Seriously. It’s very highly recommended to get a second SSD and just run
Linux natively on your PC if you want to use OSS tools to develop. Youll be
much closer to the config of the people who write those tools, and will
avoid the same traps they avoid.  This means when/if anything breaks, a
simple system update - which is and has been painlessly automated since
forever - will likely just make the problem “go away” within days to a week
or so.

If you’re used to system updates on any non-OSS system you are totally
unprepared for how important / actually effective  they really are on a
Linux distro.

Is it your computer? The only real reason for having to work always within
a VM can be because it doesn’t belong to you, and the owner doesn’t want
you to change the OS.

In which case I would point out how cheap a raspberry PI is - and note that
dispite not being an x86, many projects — including this one — will work
and build on a Debian system on arm.
Just make sure that whatever small board computer you choose, you have a
minimum 1 GB ram, or you will have trouble building yosys.

If you go that route, I’d also recommend installing the “standalone”
version of the cloud9 ide, then you can easily work over the network from
any browser. I use this kind of setup at work to embed a system with FPGA’s
in scientific laboratory equipment. (I’m a computer engineer by training,
and an Experimental Scientist by job title).

I would also recommend one of hardkernel.com  O-droid boards - and get an
SSD chip. More expensive than a raspberry pi plus sdcard, but the
performance increase - greater reliability  and “wait time reduction” -
more than compensates the price differential.

If a raspi - it is better to boot off a thumb drive as root than it is to
rely upon an sdcard. The SDcard interface sometimes corrupts drives if
booted from - it was intended for digital cameras or MP3 players, not
running an OS with random tiny io access patterns.

Anyway, good luck, and finally - try just
Plugging it into a different port. Might work again.
Also - there is a known problem if you have more than one ftdi chip USB
device plugged in on windows. Sometimes breaks iceprog access.

jasonlee1001, 2019-03-31:
> It’s the paid add on to virtualbox, but you can trial it. Otherwise you could get a thumb drive of a 16GB or larger size, put Ubuntu on it, boot off it and try icestorm from within the livebooted environment. It’ll be slow (unless it’s a really fast “USB ssd”) but it very probably will work, because native. Or, you could try my instructions at https://github.com/RGD2/icestorm-git-sing Which is how I automate building / using icestorm on windows with virtualbox, vagrant and Singularity. (Makes it very easy to delete/replace the VM, but you might hit the same issue on your hardware). I’ve done a fair bit of low level USB device development on windows, and I can report that it stinks vs any other platform. Using a VM doesn’t mean you aren’t using MSFT’s fairly crummy/lazily written USB implementation, and there are many additional issues with it that just don’t exist elsewhere. Issues like: move the device to a different port, and it will start all over again with driver discovery/installation, like its never seen the device before. You can quite easily have multiple different version drivers installed *on different ports* because you moved it to a second port, and “updated” the driver at some point. No other OS does this, and it just creates drama. On Linux, by comparison, you can just tell it what driver or even actions to take when it sees a particular vid/pid at any port. Or if you really want you could reimplement per-port decisions like windows too - they’re not the default though. Most annoying is how long the “oh look at me, working so hard finding a Driver for you” phase windows takes when new (especially USB default profile) devices are first attached. Every competent OS just works - the whole point of the USB standard is to predefine drivers so thing like mice, or USB-serial or USB audio, etc etc can “just work”. But MS sat on that standard board, and made sure that their preference of taking zero responsibility or effort to write drivers for their platform was a-ok too. Seriously. It’s very highly recommended to get a second SSD and just run Linux natively on your PC if you want to use OSS tools to develop. Youll be much closer to the config of the people who write those tools, and will avoid the same traps they avoid. This means when/if anything breaks, a simple system update - which is and has been painlessly automated since forever - will likely just make the problem “go away” within days to a week or so. If you’re used to system updates on any non-OSS system you are totally unprepared for how important / actually effective they really are on a Linux distro. Is it your computer? The only real reason for having to work always within a VM can be because it doesn’t belong to you, and the owner doesn’t want you to change the OS. In which case I would point out how cheap a raspberry PI is - and note that dispite not being an x86, many projects — including this one — will work and build on a Debian system on arm. Just make sure that whatever small board computer you choose, you have a minimum 1 GB ram, or you will have trouble building yosys. If you go that route, I’d also recommend installing the “standalone” version of the cloud9 ide, then you can easily work over the network from any browser. I use this kind of setup at work to embed a system with FPGA’s in scientific laboratory equipment. (I’m a computer engineer by training, and an Experimental Scientist by job title). I would also recommend one of hardkernel.com O-droid boards - and get an SSD chip. More expensive than a raspberry pi plus sdcard, but the performance increase - greater reliability and “wait time reduction” - more than compensates the price differential. If a raspi - it is better to boot off a thumb drive as root than it is to rely upon an sdcard. The SDcard interface sometimes corrupts drives if booted from - it was intended for digital cameras or MP3 players, not running an OS with random tiny io access patterns. Anyway, good luck, and finally - try just Plugging it into a different port. Might work again. Also - there is a known problem if you have more than one ftdi chip USB device plugged in on windows. Sometimes breaks iceprog access.
> -- -- Remy

Thank you for very very detail explanations.
I will try some suggested cases that you mention about.

mmicko, 2021-09-06:
Closing due to be VM related issue

closed 2021-09-06 by mmicko

jasonlee1001, 2021-11-10:
> Hi @jasonlee1001, today i had the same problem. I use POP OS 21.04 (based on UBUNTU) on VirtualBox 6.1 (windows 10), and when I load a program i get the same write error.
> 
> Did you find a solution to the problem?
> 
> > Hi, I use Ubuntu 16.04 on VirtualBox (host system is Windows 7) The board I use is iCEstick Evaluation Kit. After installing all tools and the command:
> > ```
> >      cd /icestorms/examples
> >      make
> >      iceprog example.bin
> > ```
> > 
> > 
> >     
> >       
> >     
> > 
> >       
> >     
> > 
> >     
> >   
> > The program replys:
> > > init..
> > > cdone: high
> > > reset..
> > > cdone: low
> > > flash ID: 0x20 0xBA 0x16 0x10 .....
> > > file size: 32220
> > > erase 64kB sector at 0x000000..
> > > programming..
> > > Write error (single byte, rc=-110, expected 1).
> > > ABORT.
> > 
> > 
> > Can anyone know how to solve this ?

I think maybe you need to install Virtual Box Extension Pack,
please refer to: https://www.virtualbox.org/wiki/Downloads

Support for USB 2.0 and USB 3.0 devices, ......