YosysHQ/icestorm #210: Errors in nxtPNR

jrlawton, 2019-04-02:
Hi,
installing on a Kubuntu 18.04 machine I get fatal errors as follows:

john@Ada:~/icestorm/arachne-pnr/nextpnr$ cmake -DARCH=ice40 -DCMAKE_INSTALL_PREFIX=/usr/local .
-- The C compiler identification is GNU 7.3.0
-- The CXX compiler identification is GNU 7.3.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Found PythonInterp: /usr/bin/python3 (found suitable version "3.6.7", minimum required is "3.5") 
-- Found PythonLibs: /usr/lib/x86_64-linux-gnu/libpython3.6m.so (found suitable version "3.6.7", minimum required is "3.5") 
-- Looking for pthread.h
-- Looking for pthread.h - found
-- Looking for pthread_create
-- Looking for pthread_create - not found
-- Looking for pthread_create in pthreads
-- Looking for pthread_create in pthreads - not found
-- Looking for pthread_create in pthread
-- Looking for pthread_create in pthread - found
-- Found Threads: TRUE  
-- Boost version: 1.65.1
-- Found the following Boost libraries:
--   filesystem
--   thread
--   program_options
--   iostreams
--   system
--   chrono
--   date_time
--   atomic
--   regex
-- Found OpenGL: /usr/lib/x86_64-linux-gnu/libOpenGL.so   
CMake Warning at /usr/share/cmake-3.10/Modules/FindBoost.cmake:1626 (message):
  No header defined for python-py367; skipping header check
Call Stack (most recent call first):
  CMakeLists.txt:139 (find_package)

CMake Warning at /usr/share/cmake-3.10/Modules/FindBoost.cmake:1626 (message):
  No header defined for python-py36; skipping header check
Call Stack (most recent call first):
  CMakeLists.txt:146 (find_package)

CMake Error at CMakeLists.txt:192 (find_package):
  Could not find a package configuration file provided by "Eigen3" with any
  of the following names:

    Eigen3Config.cmake
    eigen3-config.cmake

  Add the installation prefix of "Eigen3" to CMAKE_PREFIX_PATH or set
  "Eigen3_DIR" to a directory containing one of the above files.  If "Eigen3"
  provides a separate development package or SDK, be sure it has been
  installed.

-- Configuring incomplete, errors occurred!
See also "/home/john/icestorm/arachne-pnr/nextpnr/CMakeFiles/CMakeOutput.log".
See also "/home/john/icestorm/arachne-pnr/nextpnr/CMakeFiles/CMakeError.log".

[CMakeOutput.log](https://github.com/cliffordwolf/icestorm/files/3034189/CMakeOutput.log)
[CMakeError.log](https://github.com/cliffordwolf/icestorm/files/3034190/CMakeError.log)

I'm not sure how to proceed from here, help would be appreciated.

John

daveshah1, 2019-04-02:
You will need to install `libeigen3-dev`, this is a new dependency from YosysHQ/nextpnr#219

I'll submit a PR to update the instructions in a moment.

jrlawton, 2019-04-02:
Thanks, that fixes the problem.

closed 2019-04-02 by jrlawton