YosysHQ/icestorm #213: Write error (chunk, rc=-1, expected 256)

pcossutta, 2019-04-28:
Hi,

I'm working on OSX and my first install was through Apio but I get the same result if I build from source.
 
Data is written to the flash but an error is generated on the verification.

The log is as following:

> init..
> cdone: high
> reset..
> cdone: low
> flash ID: 0xEF 0x40 0x16 0x00
> file size: 104090
> erase 64kB sector at 0x000000..
> erase 64kB sector at 0x010000..
> programming..
> reading..
> Write error (chunk, rc=-1, expected 256).
>
>ABORT.

I modified iceprog to read only 128 bytes at a time and it works properly. I suspect a problem on libusb but couldn't track it down. I tried with both, libftdi and libftdi0 (installed by brew) and the result is the same.

Does anyone knows what is wrong ?. Does iceprog works on OSX on others machines ?

I can fix the problem and if anyone is interested I could add a long option to the command line to reduce the read_chunk size. I've already proposed the -X switch to not perform the validation, maybe another letter to reduce de size of read operations or a long option could be used.

By the way, I checked buffer sizes on libftdi and they seem to be OK.

Thanks.

cliffordwolf, 2019-04-29:
CC @mmicko

mmicko, 2019-04-29:
Just tried on my machine and it works fine (this is programming Upduino2 board)

iceprog hardware.bin 
init..
cdone: high
reset..
cdone: high
flash ID: 0xEF 0x40 0x16 0x00
file size: 104090
erase 64kB sector at 0x000000..
erase 64kB sector at 0x010000..
programming..
reading..
VERIFY OK
cdone: high
Bye.

I am on 17.6.0 kernel (macOS High Sierra 10.13.5)  and clang --version returns 9.1.0
Can you please send me output of your uname -a and clang --version ? 
Wonder if there is some kernel issue related to usb.

pcossutta, 2019-05-03:
Sorry for the delay. I'm using macOS Mojave 10.14.4 (kernel 18.5) and clang --version returns 10.0.1.

I checked the libusb github and there seems to be an issue that could be correlated with this (Its not exactly the same). I'll try to compile libusb on the weekend (instead of installing it by brew) and let you know what happen.

pcossutta, 2019-10-02:
Finally I could track the bug and it was a hardware one due to a wrong solder on the 12Kohm resistor connected to the REF pin on the FTDI. There is nothing wrong with the toolchain. Sorry for the noise and thank you for all your support.

For the record, when the resistor from REF to GND is missing, if transfers are done with chunks of 128 bytes it works, when done in 256 bytes it doesn't.

closed 2019-10-02 by pcossutta