YosysHQ/icestorm #217: Create a release?

FFY00, 2019-05-25:
Could you please consider creating a release so that this software is suitable for distribution?

FFY00, 2020-03-28:
Friendly ping @cliffordwolf

FFY00, 2020-06-26:
Sorry to bother, any news on this?

RomaVis, 2022-09-16:
This would require icestorm maintainers to have some sort of release schedule, like it's currently done for yosys and nextpnr. That's convenient for end users, but will it bring any actual improvement?

I'm not a maintainer here, so it's just my own impression, but anyway:
1. Icestorm doesn't seem to undergo many changes these days, so perhaps each new commit on master can be considered a new "release"?
2. There won't be much difference from using latest master compared to using hand-made release tag. From what I see there's no test suite of any kind, so it's not realistically possible to validate all the code and say "good to be released". Even if there are release tags, they would offer same level of stability as any commit you take from master.
3. Not having releases doesn't make it impossible for distro maintainters to package icestorm - they can tag package with timestamp instead of a version. One example where this is done is _salfter_ overlay for Gentoo: [link](https://github.com/gentoo-mirror/salfter/tree/master/sci-electronics/icestorm). Distros can release packages at their own schedule (at today's pace it's not impossible to have new package revision for every new commit on master branch).

One benefit of adding releases is that versioning scheme can be used to indicate loss of backward compatibility in the interface/API. Does icestorm have any interface important for end users besides CLI options?

Related issues:
#163 #280

dseight, 2023-06-25:
For example, absence of a tagged releases makes it impossible to add icestorm to the macOS homebrew repo (https://docs.brew.sh/Formula-Cookbook#basic-instructions):

> Before submitting a new formula make sure your package:
> ...
> - has a stable, tagged version (i.e. isn’t just a GitHub repository with no versions)

So, for now it's just impossible to make this tools easily accessible for the macOS users.