YosysHQ/icestorm #225: iceprog does not find iCEstick on Ubuntu

CristinaZN, 2019-07-05:
when i used iceprog xxxx.bin, it printed message like this:
"Can't find iCE FTDI USB device (vendor_id 0x0403, device_id 0x6010 or 0x6014)."
and I've already created the file  /etc/udev/rules.d/53-lattice-ftdi.rules

matthiasbock, 2019-07-05:
That sounds like your USB device wasn't detected. Please run
~~~
lsusb
~~~
in a terminal to list all USB devices connected to your computer. In the resulting list check, whether the iCEstick is listed (Look for the keywords Lattice, FTDI, FT2232). If it is listed but the Vendor or Product IDs differ from the above, please post the output of lsusb here. If it is not listed, try:
~~~ 
sudo tail -fn0 /var/log/syslog
~~~
to monitor syslog messages. Then disconnect the iCEstick from und reconnect it to the USB port. The output on the syslog should indicate, whether the device was detected or not and whether the USB communication with the FTDI chip works. Is anything happening on the syslog, when you plug in the iCEstick? Are there any error messages?

CristinaZN, 2019-07-07:
thanks a lot, the problem is perfectly solved!

closed 2019-07-07 by CristinaZN