YosysHQ/icestorm #228: icebox_vlog error decoding first byte

edsonms, 2019-07-29:
I keep getting this error when trying to reverse engineering ts4900 bitstream file:
[image]

The bitstream file is the following:

[ts4900-fpga.bin.zip](https://github.com/cliffordwolf/icestorm/files/3443646/ts4900-fpga.bin.zip)

Any idea of what it might be?

daveshah1, 2019-07-30:
`icebox_vlog`, and all the other `icebox_` tools, need an unpacked .asc bitstream file, not a raw .bin file. Use `iceunpack` to convert between the two.

edsonms, 2019-07-30:
> `icebox_vlog`, and all the other `icebox_` tools, need an unpacked .asc bitstream file, not a raw .bin file. Use `iceunpack` to convert between the two.

When I try to use iceunpack I get an error "Segmentation fault". I tried it on ubuntu and OSX, same error. It generates an output, but if I then try to use icebox_vlog to convert to verilog it does'nt work and give me more errors:
[image]
[files_generated.zip](https://github.com/cliffordwolf/icestorm/files/3447932/files_generated.zip)

edsonms, 2019-08-05:
@cliffordwolf could you please give me some help on solving these errors?

matthiasbock, 2019-08-06:
@edsonms, for clarification, you are referring to the following board, correct?

https://wiki.embeddedarm.com/wiki/TS-4900

The page mentions, that the "OS booloader" takes care of bitstream loading. It may be using a different file format. A simple way to check this would be to compare the file size of your binary file to the size of a raw bitstream file.

cliffordwolf, 2019-08-08:
6943ad7 fixes the issue in iceunpack and 2ccae0d fixes a related issue in icebox_vlog. Now decoding that file and converting it to verilog seems to work just fine.

closed 2019-08-08 by cliffordwolf

edsonms, 2019-08-08:
@cliffordwolf sorry to bother you and sorry my ignorance (I am trying to learn more about your icestorm tools), but even after your recent commits it seems to me I still can't iceunpack that binary:
[image]

Sorry again

edsonms, 2019-08-08:
The error reported at the end is: "Segmentation fault: 11"

cliffordwolf, 2019-08-09:
That just looks like the old behavior before 6943ad7. Are you sure you are using the new version?

edsonms, 2019-08-09:
> That just looks like the old behavior before [6943ad7](https://github.com/cliffordwolf/icestorm/commit/6943ad7afae03fb19f55522d0e22f2db9b25bb95). Are you sure you are using the new version?

@cliffordwolf Yes, as you can see from the screenshot below:
[image]
[image]

daveshah1, 2019-08-09:
Are you sure you have actually run `make install`?

edsonms, 2019-08-09:
> Are you sure you have actually run `make install`?

Oh gosh, my fault. Sorry about that, beginner mistake. I have succesfully unpacked, and now I am trying to use icebox_vlog. 

Thanks for the support

edsonms, 2019-08-09:
Everything has worked after the corrections made. Thank you all for the support.

edsonms, 2019-08-12:
@cliffordwolf is it normal that the icebox_vlog generates verilog code with wires declarations using the same name as the input/output module ports?
[image]
[image]

commit 2ccae0d3 (2019-08-08): Only dump memory initialization in icebox_vlog if present in ASC file, fixes #228
commit 6943ad7a (2019-08-08): Only write bram data to ASCII output if bram data is present, fixes #228