YosysHQ/icestorm #231: Error: Got .dsp0_tile statement for ipcon tile 0 7730736.

xobs, 2019-08-20:
I'm getting the above error whenever I try to run `icepack` on Windows.

```sh
PS> icepack build/blink.asc blink.bin
Error: Got .dsp0_tile statement for ipcon tile 0 7535064.
PS>
```

I'm running icestorm cfac9f950412ae139fa3c085c21881cbaf24822f (which I'm packaging up at https://github.com/xobs/toolchain-icestorm/releases ).  This doesn't happen when I use the `icepack.exe` from 5ab07ed32a768d484284f1d0e58f61c2ef9d398a (as packaged by https://github.com/FPGAwars/toolchain-icestorm/releases ) with the exact same `blink.asc` file.

xobs, 2019-08-20:
It should be noted that this happens regardless of the design I use -- It's happening on everything from a simple "blink" to a full SoC.  None of these use DSPs.

daveshah1, 2019-08-20:
It would be handy to have the blink.asc file too just to check.

xobs, 2019-08-20:
Github won't let me attach `blink.asc` here, so here's a gist with it: https://gist.github.com/xobs/086a1e85de3a5022f936f00c750645e0

daveshah1, 2019-08-20:
Seems like some very odd `istream` behaviour - afaics it must be coming from this line https://github.com/cliffordwolf/icestorm/blob/master/icepack/icepack.cc#L770. Looks like that `blink.asc` has Unix line endings (unless mangled by gist), not sure how a native Windows istream handles that.

xobs, 2019-08-20:
That's a good idea, but it looks like gist is mangling the line endings.  vim indicates it's a dos file, as generated by nextpnr-ice40.  If I convert the line endings to unix and re-run icepack, there is no change.

cliffordwolf, 2019-08-20:
GitHub does allow posting .zip files. So just zipping it would probably be the easiest way of sharing that file exactly as it is.