YosysHQ/icestorm #232: Icestrom installation issue

AEW2015, 2019-08-21:
with : cmake -DARCH=ice40 -DCMAKE_INSTALL_PREFIX=/usr/local .

CMake Error at CMakeLists.txt:189 (find_package):
  Could not find a package configuration file provided by "Eigen3" with any
  of the following names:

    Eigen3Config.cmake
    eigen3-config.cmake

  Add the installation prefix of "Eigen3" to CMAKE_PREFIX_PATH or set
  "Eigen3_DIR" to a directory containing one of the above files.  If "Eigen3"
  provides a separate development package or SDK, be sure it has been
  installed.

AEW2015, 2019-08-21:
My solution:
https://answers.ros.org/question/253181/howto-eigen-installation/

xobs, 2019-09-07:
You should be able to do `apt-get install libeigen3-dev` and then it will just work.  This is what we do to build the Fomu toolchain for Raspberry Pi.

closed 2021-09-06 by mmicko