YosysHQ/icestorm #234: Crashes after (successfully) programming

emilazy, 2019-08-28:
I haven't done any investigation as to the cause yet, but in case it's a known bug I figured I'd report it; this consistently happens for me on Linux with an HX8K breakout board:

```
[emily@yukari:~/src/fpga-playground/blinky-nmigen]$ sudo iceprog -S result/top.bin
Please touch the device.
init..
cdone: low
reset..
cdone: low
programming..
cdone: high
Bye.
*** buffer overflow detected ***: iceprog terminated
Aborted
```

cliffordwolf, 2019-09-01:
I tried running `iceprog` in valgrind, but don't get an error. (I'm on Ubuntu 18.04.3 LTS.) No idea what's going on here. My guess would be `atextit()` handlers being run in a different order on my system and yours.

Can you create a stack backtrace for this and post it here?

emilazy, 2019-09-02:
Annoyingly, the segfault seems to be in libftdi:

```
*** buffer overflow detected ***: /home/emily/src/icestorm/iceprog/iceprog terminated

Thread 1 "iceprog" received signal SIGABRT, Aborted.
0x00007ffff7a60be0 in raise () from /nix/store/g88xj01xv7xzdx9dy2zwga0vk56fpfbm-glibc-2.27/lib/libc.so.6
(gdb) bt
#0  0x00007ffff7a60be0 in raise () from /nix/store/g88xj01xv7xzdx9dy2zwga0vk56fpfbm-glibc-2.27/lib/libc.so.6
#1  0x00007ffff7a61dc1 in abort () from /nix/store/g88xj01xv7xzdx9dy2zwga0vk56fpfbm-glibc-2.27/lib/libc.so.6
#2  0x00007ffff7aa22ac in __libc_message () from /nix/store/g88xj01xv7xzdx9dy2zwga0vk56fpfbm-glibc-2.27/lib/libc.so.6
#3  0x00007ffff7b2ef68 in __fortify_fail_abort () from /nix/store/g88xj01xv7xzdx9dy2zwga0vk56fpfbm-glibc-2.27/lib/libc.so.6
#4  0x00007ffff7b2ef81 in __fortify_fail () from /nix/store/g88xj01xv7xzdx9dy2zwga0vk56fpfbm-glibc-2.27/lib/libc.so.6
#5  0x00007ffff7b2d070 in __chk_fail () from /nix/store/g88xj01xv7xzdx9dy2zwga0vk56fpfbm-glibc-2.27/lib/libc.so.6
#6  0x00007ffff7b2ee87 in __fdelt_warn () from /nix/store/g88xj01xv7xzdx9dy2zwga0vk56fpfbm-glibc-2.27/lib/libc.so.6
#7  0x00007ffff7e324a0 in _usb_async_cleanup () from /nix/store/ppqm9bwznn6lx18pwzrsk4iiaswp6yd2-libftdi-0.20/lib/libftdi.so.1
#8  0x00007ffff7e3397a in ftdi_usb_close () from /nix/store/ppqm9bwznn6lx18pwzrsk4iiaswp6yd2-libftdi-0.20/lib/libftdi.so.1
#9  0x000000000052f110 in mpsse_close () at mpsse.c:361
#10 0x000000000052b49c in main (argc=-17319724, argv=0x616000000080) at iceprog.c:975
```

I'm on NixOS unstable (which is a pretty weird environment, but that shouldn't matter too much here), using libftdi 0.20, and I'm programming an HX8K breakout board.

emilazy, 2019-09-02:
Okay, maybe NixOS is actually to blame here; it has both `libftdi` at 0.20 and `libftdi1` at 1.4, and the segfault doesn't occur with the latter. I guess I'll update the icestorm package to use the newer version.

emilazy, 2019-09-02:
This seems to be an [old, known bug in libftdi](http://libftdi.141977.n3.nabble.com/how-to-ftdi-usb-close-without-segfault-td3202143.html). Not sure why nixpkgs has the old version packaged, but this probably isn't an icestorm bug.

closed 2019-09-02 by emilazy