YosysHQ/icestorm #237: Wrong PLL frequency in PHASE_AND_DELAY mode on UP5

HansHamster, 2019-09-23:
Minimal example to show the problem:

`module top(
  output clk_out
);

wire clk_48M;

SB_HFOSC inthosc (
  .CLKHFPU(1'b1),
  .CLKHFEN(1'b1),
  .CLKHF(clk_48M)
);

defparam inthosc.CLKHF_DIV = "0b00";

wire clk1, clk2;

// icepll -i 48 -o 32 -S
SB_PLL40_2F_CORE #(
  .FEEDBACK_PATH("PHASE_AND_DELAY"),
  .DIVR(4'b0010),	// DIVR =  2
  .DIVF(7'b0000001),	// DIVF =  1
  .DIVQ(3'b101),	// DIVQ =  5
  .FILTER_RANGE(3'b001)	// FILTER_RANGE = 1
) uut (
  .LOCK(),
  .RESETB(1'b1),
  .BYPASS(1'b0),
  .REFERENCECLK(clk_48M),
  .PLLOUTGLOBALA(clk1),
  .PLLOUTGLOBALB(clk2)
);

assign clk_out = clk1;

endmodule
`

The PLL parameters are calculated for a 48MHz input and a 32MHz output with `icepll -i 48 -o 32 -S`.
Connecting each clock to the output, I measure 48MHz on clk_48M, but 80MHz on clk1 and clk2 instead of 32MHz.
According to TN1251, the output clock should be Refclk*(DIVF+1)/(DIVR+1), so 48MHz*2/3=32MHz, but for some reason the actual frequency is much higher.

daveshah1, 2019-09-23:
I am not 100% sure where the factor of 2.5 is coming from, previously I have seen factors of either 4 or 7 depending on SHIFTREG_DIV_MODE (see #198)

smunaut, 2020-02-12:
I would check that the PLL is locked at all ...

HansHamster, 2020-03-05:
Finally had time to do some more digging.
Part of the issue was, that I simply forgot the
`  .PLLOUT_SELECT_PORTA("SHIFTREG_0deg"),
  .PLLOUT_SELECT_PORTB("SHIFTREG_90deg")`
which brings the frequency down to 20 MHz.

I then compared the PLL parameters calculated by icepll and iceCube/Radiant and it looks like (at least for the cases I tried) that all parameters are identical except for DIVQ.
If I set DIVQ to 3 as calculated by iceCube instead of 5, I get the expected 32 MHz.