YosysHQ/icestorm #238: Makefile assumes python binary is `python3`

xobs, 2019-10-22:
The icebox `Makefile` calls `python3`, which is only the name of this program on Unix-like systems (according to PEP 394).  The official executable name on other platforms, notably on Windows, is `python`, and so the build fails on those platforms.

xobs, 2019-10-22:
It's easy enough to add a variable to `config.mk` that lets you specify what python is called, which can get `icebox` to compile.  However, how much updating should we do?

Should `icefuzz` be updated to use $(PYTHON)?  What about `icetime`?  And what about various scripts such as `mktest.py`?

mmicko, 2021-09-06:
Your fix have been merged, note change from PYTHON to PYTHON3

closed 2021-09-06 by mmicko