YosysHQ/icestorm #243: g++-8.3.0 coredumps when building icestorm

RobertBaruch, 2019-12-23:
This is just a heads-up. I don't really want to take the time to debug this further, since it's fairly straightforward to install g++-7, temporarily downgrade (via `sudo update-alternatives`), and just use that. Still, here is the error:

```
python3 timings.py up5k > timings-up5k.cc                                                                               
g++ -MD -O2  -Wall -std=c++11 -I/usr/local/include -DPREFIX='"/usr/local"' -DCHIPDB_SUBDIR='"icebox"'   -c -o timings-up5k.o timings-up5k.cc                                                                                                    
g++: internal compiler error: Segmentation fault signal terminated program cc1plus                                      
Please submit a full bug report,                                                                                        
with preprocessed source if appropriate.                                                                                
See <file:///usr/share/doc/gcc-8/README.Bugs> for instructions.                                                         
make[1]: *** [<builtin>: timings-up5k.o] Error 4
```

smunaut, 2020-02-19:
Are you sure it's not just running out of RAM ?  How much RAM does that machine have ?

I have gcc 8.3.0 (but on another distro) and it built fine.

emilazy, 2020-02-19:
In any case, a g++ segfault would be GCC's bug, not IceStorm's (though it might be worth working around if other people experience it too).

SmallRoomLabs, 2020-03-04:
Got the same on `g++ (Ubuntu 7.4.0-1ubuntu1~18.04.1) 7.4.0` building in a docker container.

[image]

smunaut, 2020-03-04:
Check dmesg ... my guess is you're running out of RAM.
You need several Gig of ram to build that one file.

SmallRoomLabs, 2020-03-04:
Yes, that was it.  I upped the ram to 8 gigs and and it all compiled just fine.  Strange that an oom-condition results in an internal error instead of a proper error message in gcc.

smunaut, 2020-03-04:
That's just the way linux works, by default overcommit is enabled. So when you do a malloc you never get NULL you always get a valid pointer but when you try to use it, if the kernel can't actually find any memory ... it SIGKILLs the process.  You can google "linux memory overcommit" for details.

SmallRoomLabs, 2020-03-05:
Oh, that's a horrible concept (in general). Once upon a time when I learned C on PDPs and VAXens allocs actually did what they should. But as far as I remember I've got "correct" behavior in Linux too, but that was probably back in the early kernels in the mid to late 1990s.

So this means that I can stop checking malloc/callocs for null in my code (not that I do much of them in my embedded programming anyways) then and just trust that the magic goblins in the kernel will be able to give me the memory sometime in the future anyways.    ;-)

mmicko, 2021-09-06:
Closing, due to be memory related issue and not icestorm related

closed 2021-09-06 by mmicko